HASwitchPlate/openHASP #1: Open to Pull Requests?

Desterly, 2020-03-23:
Less of an issue but more of a question.

I see in your project listing the items you'd like to add of which a few I've got working in a fork.

*   spiffs file handler working
*   saving calibration data
*   showing images

Are you open to pull requests for those?

fvanroie, 2020-03-23:
The project is very much ongoing and any help is much appreciated.
Do you have a link to the fork? Or feel free to submit a pull request.

Desterly, 2020-03-30:
Sorry for the delay.

I am rebasing my updates on 0.0.11 and cleaning out the test data.   Opening pull requests as I clean these out.  If there's better ways to handle it based on your coding preferences, than feel free to simply consider them suggestions

closed 2020-05-04 by fvanroie