HASwitchPlate/openHASP #2: Instructions need some details

zenonasz, 2020-03-28:
i have build and flashed your code but i get nothing in the display. i used the same libraries and configuration and i was able to run littlevgl with tft_espi just fine.  looks something is missing from the project or instructions are not clear.
Can for example be used for integration with home assistant in the same way original HAPlate is used, throught "first time setup automation"?
Is it mandatory to upload fonts?

fvanroie, 2020-03-28:
Hello zenonasz, thanks for trying hasp-lvgl. The project is still in the development phase and changing rapidly. The documentation is also *under construction*. Your input can help us update the missing parts.

You should certainly be able to use a TFT_eSPI compatible display, however hasp-lvgl does *not* use the User_Setup.h file of the library directly. All the TFT_eSPI related parameters are defined using `build_flags` in the platformio.ini file. You need to copy the parameters from User_Setup.h into the appropriate platformio.ini section.

Can you please provide the following information:
- Model of ESP and TFT display
- Version (branch) of hasp-lvgl you are building
- The User_Setup.h parameters you used for the previous project

This way we can identify the issue in the configuration.

You are not required to upload fonts, the built-in font will be used of no fonts are uploaded.

It is possible to setup the layout of the screen from HA or OpenHab, but no automations are provided yet. There are some users working on that part and it will be included on github at a later stage.

You can also reach out on our discord channel https://discord.gg/dzmkbb

zenonasz, 2020-03-28:
Thanks for the prompt response @fvanroie ,

Thanks for the effort porting this to open hardware and ditch out the nextion thing :). glad you initiate it, i am willing to contribute.
 I am an experienced developer not so much with C and C++, it my first project on platformio tbh but i have worked with various IDE s for embedded development other than arduino (eclipse with esp_sdk and sming framework for esp8266).

pretty much i get this up and running but i get a lot of "Out of memory" from littlvgl components. 
`pio/libdeps/esp32dev/lvgl/src/lv_misc/lv_mem.c@182->Couldn't allocate memory` i have tried to the pages.jsonl example you have on the wiki. i

- i am using TTGo ESP32 with a Non arduino display (its Adafruit RPi touch 2.8)
-i am using your default branch 0.11
-i have understood how platformio uses its build flags. However even when i define it with 
` -D USER_SETUP_LOADED
    -D ILI9341_DRIVER` i get wrong report driver in the logs, ` Display driver = 37697` also processor not makes sense `Processor    = ESP50`

The display i use it basically uses another driver `Adafruit_FT6206` which tft-espi is not supporting. however i did some tweaking on hasp_gui and it seems to work fine.

Now about the pages and the fonts...

Pages, can i configure pages through mqtt istead of the json file. if so we can still hasp HA automatic integration to do it. 

Thanks again, 
Willing to contribute and undertake a task, let me know

Regards from sunny, yet locked down Cyprus

fvanroie, 2020-03-29:
Thanks, your feedback and help is much appreciated.
To check the available memory you can check the Information page in the web interface:

[image]

Start out with a minimal jsonl and see if that solves the memory errors. You can also send jsonl objects to the `hasp/[name]/command/jsonl` mqtt topic. This has the same effect as putting it in a pages.jsonl file, but is more dynamic.

The issue with `Display driver` and `Processor` will be fixed in the upcoming release.

For fonts, you can get the latin1 and fontawesome pack from [here](https://github.com/fvanroie/HMI-Font-Pack/releases).

The idea is to make the display work similarly to the original HASwitchPlate project, but some changes need be done. I don't use HA myself so any feedback on the automations there is most welcome.

I'm curious as to what change you did to `hasp_gui` to make it work... is that specific to the screen you have?

fvanroie, 2021-01-08:
We have moved the documentation into a separate github pages repository.
The documentation link is https://fvanroie.github.io/hasp-docs/

Feel free to check it out and [let us know](https://github.com/fvanroie/hasp-lvgl/discussions) if anything is missing.

closed 2021-01-08 by fvanroie