HASwitchPlate/openHASP #10: Fail to compile -- Using the specific lvgl version to compile successfully

betamoojw, 2020-09-29:
As the lvgl is evaluating and iterating causing some incompatible issues. As stated `lib_deps =
    https://github.com/littlevgl/lvgl.git ;lvgl@6.1.2 - Not in library yet` in the platformio.ini, modify it to `https://github.com/littlevgl/lvgl.git#v6.1.2 ;lvgl@6.1.2 - Not in library yet` and then compile successfully.

fvanroie, 2020-10-12:
Which branch of hasp-lvgl did you upload?

I'm working on polishing up the repository and including a more streamlined installation, with pre-compiled binaries for ESP32/8266. So this shouldn't be an issue in the future.

betamoojw, 2020-10-14:
The branch of hasp-lvgl used in my case is the default (0.0.11).
As mentioned above, I modified `lib_deps = https://github.com/littlevgl/lvgl.git#v6.1.2  ;lvgl@6.1.2` with specified lvgl version and then compile and run it successfully on my hardware.

fvanroie, 2020-10-15:
I made the update to v6.1.2 in the 0.0.11 branch. This should fiw that complile issue.
In the near future, the new default branch will switch to 0.2.0 as that is the most recent and active branch being developed.

closed 2020-10-15 by fvanroie