HASwitchPlate/openHASP #12: No response to touch event however the touch coordination can be detected after guiShowPointer is true

betamoojw, 2020-10-12:
As mentioned those **sequenced** steps in Getting Started in Wiki the  [https://github.com/fvanroie/hasp-lvgl/wiki/Getting-Started](url), only MQTT Client is Connected then the lvgl will response to touch event (see void haspReconnect() function for the detail).

fvanroie, 2020-10-12:
Can you provide some more details about your setup?
- Which MCU and touchscreen are you using?
- Which branch of hasp-lvgl did you upload?
- Please include the pages.jsonl file you uploaded
- Can you provide a screenshot image (via webinterface or a picture) of the UI

Thanks

betamoojw, 2020-10-14:
> * Which MCU and touchscreen are you using?
    ESP-WROOM-32 ESP-32S DevKit (see the link [https://item.taobao.com/item.htm?spm=a1z09.2.0.0.18d62e8dZyhxtN&id=602686977820&_u=e8lq2g0791f](url) ) and touchscreen - 3.2inch LCD TFT with SPI interface ILI9341 and XPT2046 (see the link [https://item.taobao.com/item.htm?spm=a1z09.2.0.0.18d62e8dZyhxtN&id=624750714382&_u=e8lq2g0948e](url) ).    

> * Which branch of hasp-lvgl did you upload?
    The branch of hasp-lvgl used in my case is the default (0.0.11).

> * Please include the pages.jsonl file you uploaded
    The pages.jsonl is based on the Example file [https://github.com/fvanroie/hasp-lvgl/wiki/Page-Layout#example-file](url) with some modifications.

> * Can you provide a screenshot image (via webinterface or a picture) of the UI
[image] and the log 
[log.txt](https://github.com/fvanroie/hasp-lvgl/files/5377842/log.txt)

fvanroie, 2020-10-15:
Could you try  branch 0.2.0? This is the actively developement banch and made the jump to lvgl 7.6.
The theme will look a bit different, but touch should work fine with your hardware.

If you have Discord, feel free to join our Discord Channel: https://discord.gg/VCWyuhF

betamoojw, 2020-10-19:
Have tried branch 0.2.0, with some comments in the file /src/hasp_gpio.cpp as below and adjustments based on my hardware everything works fine.
[image]

betamoojw, 2020-10-19:
Here is a screenshot image (via the web interface or a picture) of the UI. Additional we should update the version to 
0.2.0.
[image]

closed 2020-10-19 by betamoojw

fvanroie, 2020-10-19:
Yes indeed, I will update the version for that branch to avoid confusion.
Good to know it worked for you. This will become the default branch in the near future.