HASwitchPlate/openHASP #13: Where to put new font

12christiaan, 2020-10-26:
Hi, i love your work. and start to see if i can use it for my project.
Got it working with a ESP32-CAM + 3.5" display (ILI9488) 320x480.

I can not figure out how to use a bigger font.
I got several .zi fonts and fontawesome.
i try uploading them via the GUI, but never gave me a hint on where to find them or how to use them

Please help.

fvanroie, 2020-10-27:
Hello,

Thanks for trying out hasp-lvgl. Good to hear you got it working on the ESP-CAM.
Which branch/version of hasp-lvgl did you flash to the ESP?

When uploading a file on the HASP Configuration page, it is saved on the internal filesystem of the ESP.
You should then see the available .zi fonts in the drop-down list "Default Font". You can select a font to make it the default font.

The font support is currently rather basic. Is your goal to make the default font bigger or mix several font&sizes on the layout?

12christiaan, 2020-10-27:
See details of version below.

When uploading font from gui, i see in debug (serial) the upload is okay.
but in gui the default font option remains empty.
Also after reset, the font is still empty.

-=-
HASP Version: 0.2.1014
Build DateTime: Oct 25 2020 10:07:50 CET
Uptime: 8s
Free Memory: 164.08KB
Memory Fragmentation: 32
LVGL Memory: 20.00KB
LVGL Free: 11.88KB
LVGL Fragmentation: 1

LCD Active Page: 0

MQTT Status: Connected
MQTT ClientID: plate01-5caf40

MCU Model: 2core ESP32 rev1
CPU Frequency: 240MHz
Flash Chip Size: 4.00MB
Program Size: 1.31MB
Free Program Space: 3.00MB
Core version: v3.2.3-14-gd3e562907

12christiaan, 2020-10-27:
I found a hint.
from the serial log i see.

          _____ _____ _____ _____
          |  |  |  _  |   __|  _  |
          |     |     |__   |   __|
          |__|__|__|__|_____|__|
        Home Automation Switch Plate
        Open Hardware edition v0.2.1014

[               1.586][113792/268184 57][    0/    0  0]FILE: [SUCCESS] SPI flash FS mounted
[               1.588][113792/268184 57][    0/    0  0]FILE: Listing files on the internal flash:
[               1.618][113792/268184 57][    0/    0  0]CONF: Loading /config.json
...
[               2.881][113792/185100 38][12152/12164  1]HASP: File /pages.jsonl loaded

This indicate the it did not find ANY file ? 
but later if loads the pages file okay. ?

fvanroie, 2020-10-27:
Strange. The contents of the SPIFFS partition is indeed listed at boot-time. So an empty list there indicates no files are present.
The pages.json file is optional and if it is not there it is just ignored. But on my ESP32 it shows up as a failure.
```
FILE: Listing files on the internal flash:
FILE:    * /Lato16.zi  (3770 bytes)
FILE:    * /Poppins20.zi  (2683 bytes)
FILE:    * /config.json  (538 bytes)
CONF: Loading /config.json
...
Opening pages.json from FS failed 3
Reading pages.json from FS failed 11
Closing pages.json on FS failed 11
FONT: Loaded V5 Font File: /Poppins20.zi containing 95 characters
HASP: Custom theme loaded
HASP: Non existing file /pages.jsonl
HASP: Changing page to 0
```
I haven't tested on the ESP32-CAM, maybe the flash partitions are different. Looking at your Information page above, I notice the free program space is 3 MB. It seems to me that the whole 4MB flash is partitioned for Firmware only and no SPIFFS partition is configured.

I have:
```
Flash Chip Size: 4.00MB
Program Size: 1.26MB
Free Program Space: 1.37MB
```
Which is 2x partitions of 1.37MB for programs (primary and ota) with the remainder of 1.26 MB for SPIFFS.

Do you see the SPI flash being mounted on the first log line?
```
           _____ _____ _____ _____
          |  |  |  _  |   __|  _  |
          |     |     |__   |   __|
          |__|__|__|__|_____|__|
        Home Automation Switch Plate
        Open Hardware edition v0.2.1014

FILE: [SUCCESS] SPI flash FS mounted
```
Do you have a line like this in the board.ini?
```
board_build.partitions = esp32_partition_app1300k_spiffs1216k.csv ; default.csv
```

12christiaan, 2020-10-27:
full start log.

[               1.555][113792/268328 57][    0/    0  0]Serial started at 115200 baud

           _____ _____ _____ _____
          |  |  |  _  |   __|  _  |
          |     |     |__   |   __|
          |__|__|__|__|_____|__|
        Home Automation Switch Plate
        Open Hardware edition v0.2.1014

[               1.588][113792/268328 57][    0/    0  0]FILE: [SUCCESS] SPI flash FS mounted
[               1.590][113792/268328 57][    0/    0  0]FILE: Listing files on the internal flash:
[               1.620][113792/268328 57][    0/    0  0]CONF: Loading /config.json
[               1.622][113792/267752 57][    0/    0  0]CONF: {"wifi":{"ssid":"********","pass":"********"},"mqtt":{"name":"plate01","group":"plates","host":"192.168.1.33","port":1883,"user":"********","pass":"********"},"telnet":{"enable":1},"mdns":{"enable":1},"http":{"enable":true,"port":80,"user":"","pass":""},"debug":{"teleperiod":300,"baud":11520,"host":"","port":514,"proto":0,"log":0},"gui":{"tickperiod":50,"idle1":60,"idle2":120,"bcklpin":-1,"rotation":2,"pointer":false,"calibration":[264,3520,305,3574,2]},"hasp":{"startpage":0,"startdim":80,"theme":4,"hue":106,"font":"","pages":"/pages.jsonl"}}
[               1.675][113792/267752 57][    0/    0  0]CONF: [SUCCESS] Loaded /config.json
[               1.685][113792/268432 57][    0/    0  0]Loading debug settings
[               1.695][113792/267956 57][    0/    0  0]CONF: {"teleperiod":300,"baud":11520,"host":"","port":514,"proto":0,"log":0}
[               1.709][113792/268432 57][    0/    0  0]Loading GUI settings
[               1.718][113792/267956 57][    0/    0  0]CONF: {"tickperiod":50,"idle1":60,"idle2":120,"bcklpin":-1,"rotation":2,"pointer":false,"calibration":[264,3520,305,3574,2]}
[               1.736][113792/268432 57][    0/    0  0]CONF:    * guiTickPeriod set
[               1.746][113792/268432 57][    0/    0  0]calData set [264, 3520, 305, 3574, 2]
[               1.757][113792/268432 57][    0/    0  0]Loading HASP settings
[               1.766][113792/267956 57][    0/    0  0]CONF: {"startpage":0,"startdim":80,"theme":4,"hue":106,"font":"","pages":"/pages.jsonl"}
[               1.782][113792/268432 57][    0/    0  0]CONF:    * haspStartDim set
[               1.792][113792/268432 57][    0/    0  0]CONF:    * haspThemeId set
[               1.801][113792/268432 57][    0/    0  0]CONF:    * haspThemeHue set
[               1.811][113792/268432 57][    0/    0  0]Loading WiFi settings
[               1.821][113792/267956 57][    0/    0  0]CONF: {"ssid":"********","pass":"********"}
[               1.832][113792/268432 57][    0/    0  0]Loading MQTT settings
[               1.841][113792/267956 57][    0/    0  0]CONF: {"name":"plate01","group":"plates","host":"192.168.1.33","port":1883,"user":"esp","pass":"********"}
[               1.858][113792/268432 57][    0/    0  0]Loading MDNS settings
[               1.867][113792/267956 57][    0/    0  0]CONF: {"enable":1}
[               1.876][113792/268432 57][    0/    0  0]Loading HTTP settings
[               1.886][113792/267956 57][    0/    0  0]CONF: {"enable":true,"port":80,"user":"","pass":"********"}
[               1.898][113792/268432 57][    0/    0  0]User configuration loaded
[               1.999][113792/219052 48][    0/    0  0]WIFI: Connecting to : ********
[               2.212][113792/217764 47][    0/    0  0]WIFI: Connected to SSID ********. Requesting IP...
[               2.225][113792/217916 47][    0/    0  0]WIFI: Received IP address 192.168.1.118
[               2.226][113792/218132 47][    0/    0  0]WIFI: Connected = yes
[               2.360][113792/185156 38][17400/18000  4]TFT: TFT_eSPI   : v2.2.11
[               2.361][113792/185156 38][17400/18000  4]TFT: Processor  : STM32
[               2.368][113792/185156 38][17400/18000  4]TFT: CPU freq.  : 240 MHz
[               2.378][113792/185156 38][17400/18000  4]TFT: Transactns : Yes
[               2.387][113792/185156 38][17400/18000  4]TFT: Interface  : SPI
[               2.397][113792/185156 38][17400/18000  4]TFT: Driver     : ILI9488
[               2.407][113792/185156 38][17400/18000  4]TFT: Resolution : 320x480
[               2.416][113792/185156 38][17400/18000  4]TFT: MOSI       : D13 (GPIO 13)
[               2.427][113792/185156 38][17400/18000  4]TFT: MISO       : D12 (GPIO 12)
[               2.437][113792/185156 38][17400/18000  4]TFT: SCLK       : D14 (GPIO 14)
[               2.447][113792/185156 38][17400/18000  4]TFT: TFT_CS     : D15 (GPIO 15)
[               2.457][113792/185156 38][17400/18000  4]TFT: TFT_DC     : D00 (GPIO 00)
[               2.468][113792/185156 38][17400/18000  4]TFT: TOUCH_CS   : D02 (GPIO 02)
[               2.478][113792/185156 38][17400/18000  4]TFT: Display SPI freq. : 27.0 MHz
[               2.488][113792/185156 38][17400/18000  4]TFT: Touch SPI freq.   : 2.5 MHz
[               2.498][113792/185156 38][17400/18000  4]TFT: DMA        : DISABELD
[               2.508][113792/185156 38][17400/18000  4]LVGL: Version  : 7.7.1 
[               2.518][113792/185156 38][17400/18000  4]LVGL: Rotation : 2
[               2.527][113792/185156 38][17400/18000  4]LVGL: MEM size : 20480
[               2.536][113792/185156 38][17400/18000  4]LVGL: VFB size : 32768
[               2.546][113792/185156 38][17400/18000  4]LVGL: Registering lvgl logging handler
[               2.559][113792/185108 38][17112/17152  1]Opening pages.json from FS failed 3
[... cut ...]
[               3.628][113792/161288 29][12152/12164  1]MQTT:    * Subscribed to hasp/plate01/brightness/#
[               3.640][113792/159548 28][12152/12164  1]MQTT:    * Subscribed to hasp/plate01/status
[               3.652][113792/157800 27][12152/12164  1]MQTT: binary_sensor state: [hasp/plate01/status] : OFF
[               3.663][113792/157856 27][12152/12164  1]PAGE: 0
[               3.671][113792/157856 27][12152/12164  1]HASP: Changing page to 0
[               3.683][113792/155560 26][12152/12164  1]MQTT PUB: hasp/plate01/state/page = 0

12christiaan, 2020-10-27:
Do you have a line like this in the board.ini?
board_build.partitions = esp32_partition_app1300k_spiffs1216k.csv ; default.csv
-=-
No that was NOT there. I added it, re-build and flashed.
Now it is working.
I uploaded font.zi and it appeared.

Strang that i was able to upload my own pages.jsonl file, and that works fine.
Case close.

12christiaan, 2020-10-27:
[               1.588][113792/268272 57][    0/    0  0]FILE: [SUCCESS] SPI flash FS mounted
[               1.590][113792/268272 57][    0/    0  0]FILE: Listing files on the internal flash:
[               1.726][113792/267524 57][    0/    0  0]FILE:    * /pages.jsonl  (1720 bytes)
[               1.728][113792/267084 57][    0/    0  0]FILE:    * /arial24.zi  (3454 bytes)
[               1.737][113792/267076 57][    0/    0  0]FILE:    * /config.json  (511 bytes)
[               1.760][113792/267072 57][    0/    0  0]FILE:    * /fontawesome24.zi  (120285 bytes)
[               1.795][113792/267824 57][    0/    0  0]CONF: Loading /config.json

fvanroie, 2020-10-27:
Thanks for the feedback! That is an edge-case that I haven't tested, so live and learn :)

fvanroie, 2021-01-08:
We have moved the documentation into a separate github pages repository.
The documentation link is https://fvanroie.github.io/hasp-docs/

Feel free to check it out and [let us know](https://github.com/fvanroie/hasp-lvgl/discussions) if anything is missing.

closed 2021-01-08 by fvanroie