HASwitchPlate/openHASP #14: Remove file from SPIFFS

12christiaan, 2020-10-28:
Not a issue, more a question.
While going over the code, i saw this option.
http://192.168.1.118/list?dir=
It will list all the file on the SPIFFS.
I am testing some different fonts and font size, now the SPIFFS is running full.
and the only option i found to remove them is to flash a empty image over the SPIFFS. 

did you build some other (hidden) option to remove file's?

fvanroie, 2020-10-28:
There is no _native_ file browser provided yet, as this currently is low on the priority list.

However, you can upload the 3KB `edit.htm.gz` file from the ESP32 FSBrowserrepository to SPIFFS
from: https://github.com/espressif/arduino-esp32/blob/master/libraries/WebServer/examples/FSBrowser/data/edit.htm.gz

When the  `edit.htm.gz` file is present you will see an additional File Browser button on the Main Webpage:
[image]

Using that webpage, you can right-click and delete files:

[image]

12christiaan, 2020-10-28:
Perfect, 
that works perfectly.

fvanroie, 2020-10-28:
Closing this issue and added the solution to the FAQ page of the Wiki.

closed 2020-10-28 by fvanroie