HASwitchPlate/openHASP #15: Custom font not always used?

12christiaan, 2020-10-28:
I have uploaded custom font (via GUI)
Sellected it (Via gui)
- button on page 0 does not show the new font.
- button on page 254 does show the new font.
See screen shot.
I do not understand.

log.
[               1.591][113792/268120 57][    0/    0  0]FILE: [SUCCESS] SPI flash FS mounted
[               1.593][113792/268120 57][    0/    0  0]FILE: Listing files on the internal flash:
[               1.728][113792/267372 57][    0/    0  0]FILE:    * /pages.jsonl  (1700 bytes)
[               1.733][113792/266920 57][    0/    0  0]FILE:    * /Ubuntu-Regular40.zi  (29119 bytes)
[               1.742][113792/266912 57][    0/    0  0]FILE:    * /Ubuntu-Regular32.zi  (22018 bytes)
[               1.752][113792/266920 57][    0/    0  0]FILE:    * /config.json  (540 bytes)
[               1.782][113792/266916 57][    0/    0  0]FILE:    * /fontawesome32.zi  (199639 bytes)
[               1.811][113792/267672 57][    0/    0  0]CONF: Loading /config.json
[               2.731][113792/184320 38][14804/14832  1]FONT: Loaded V5 Font File: /Ubuntu-Regular32.zi containing 224 characters
[               2.735][113792/184736 38][14548/14580  1]HASP: Custom theme loaded
[               2.787][113792/184736 38][13360/13368  1]HASP: Loading file /pages.jsonl
[               2.789][113792/184076 38][13360/13368  1]DISPATCH: jsonl
[               2.801][113792/183932 38][13232/13240  1]HASP:     * p[0].b[1] = lv_label
[               2.812][113792/183884 38][13000/13004  1]HASP:     * p[0].b[2] = lv_btn
[               2.822][113792/183900 38][12764/12768  1]HASP:     * p[0].b[3] = lv_btn
[               2.832][113792/183900 38][12528/12532  1]HASP:     * p[0].b[4] = lv_btn
[               2.842][113792/183900 38][12292/12296  1]HASP:     * p[0].b[5] = lv_btn
[               2.851][113792/183932 38][12032/12076  1]HASP:     * p[1].b[1] = lv_slider
[               2.861][113792/183932 38][11856/11876  1]HASP:     * p[1].b[3] = lv_gauge
[               2.870][113792/183932 38][11704/11712  1]HASP:     * p[1].b[4] = lv_bar
[               2.881][113792/183932 38][11588/11588  0]HASP:     * p[1].b[5] = lv_linemeter
[               2.890][113792/183932 38][11588/11588  0]HASP: Parent ID p[1].b[5] found
[               2.901][113792/183932 38][11456/11460  1]HASP:     * p[1].b[10] = lv_label
[               2.913][113792/183932 38][11316/11316  0]HASP:     * p[2].b[1] = lv_cpicker
[               2.924][113792/183932 38][11168/11168  0]HASP:     * p[2].b[2] = lv_cpicker
[               2.935][113792/183884 38][10968/10968  0]MQTT: Not connected
[               2.939][113792/183884 38][10968/10968  0]HASP:     * p[3].b[1] = lv_dropdown
[               2.968][113792/183868 38][10440/10480  1]MQTT: Not connected
[               2.970][113792/183868 38][10440/10480  1]HASP:     * p[3].b[2] = lv_roller
[               3.001][113792/183932 38][10200/10212  1]HASP:     * p[254].b[40] = lv_btn
[               3.014][113792/183932 38][ 9928/ 9944  1]HASP:     * p[254].b[50] = lv_btn
[               3.026][113792/183932 38][ 9676/ 9688  1]HASP:     * p[254].b[60] = lv_btn
[               3.028][113792/184736 38][ 9676/ 9688  1]HASP: File /pages.jsonl loaded
[               3.037][113792/184736 38][ 9676/ 9688  1]HASP: Changing page to 0

12christiaan, 2020-10-28:
Ahh, it is related to the theme.
When selecting the "Material light" the font IS used.
[image]
I will check tomorrow on how to use these theme's and if i can create my own.

fvanroie, 2020-12-13:
I will close this issue due to inactivity. If you still experience this issue, feel free to comment below.

closed 2020-12-13 by fvanroie