HASwitchPlate/openHASP #16: Text in Dropdown List not working.

12christiaan, 2020-10-31:
I have a dropdown list configured. like this.

{"comment":"---------- Page 2 ----------"}
{"objid":50,"id":6,"page":2,"x":10,"y":150,"w":150,"txt":"\uf007 Option A\n\uf007 Option B\n\uf007 Option C"}

I was expecting the text in drop down to be in TXT, but it always show Option 1, Option 2, Option 3.
[image]

arovak, 2020-10-31:
That is because the response will be in the `txt` field when you select an option on the display like this `{"p[2].b[6].txt":" Option B"}`
To setup the dropdown list use `options` instead of `txt`

`{"objid":50,"id":6,"page":2,"x":10,"y":150,"w":150,"options":"\uf007 Option A\n\uf007 Option B\n\uf007 Option C"}`

[image]

fvanroie, 2020-11-01:
Indeed, it should be `options` instead of `txt` when defining the list.
The example `pages.jsonl on the wiki was not correct. I have updated the wiki page.

fvanroie, 2020-11-15:
@12christiaan Does it work when you use the `options` parameter instead of `txt`?

fvanroie, 2020-12-12:
I will close this issue due to inactivity. If you still experience this issue, feel free to comment below.

closed 2020-12-12 by fvanroie