HASwitchPlate/openHASP #38: Reorder object IDs and implementation status

fvanroie, 2021-01-05:
The object IDs currently are arbitrary and it would help to make better sense to group them according to their purpose. Here is a proposal to re-arrange the object IDs.

**Breaking change: This will require the existing jsonl to be updated with the new IDs.**

 Controls | **id** | **newid** | **implemented** | **note**
----------|--------|:---------:|:---------------:|:---------:
OBJECT    | 91 | ?   | :heavy_check_mark:
BUTTON    | 10 | 10  | :heavy_check_mark:
BTNMATRIX | 13 | 11  | :clock9:
IMGBTN    | 14 | 12  | :x:
CHECKBOX  | 11 | 13  | :heavy_check_mark: 
SWITCH    | 40 | 14  | :heavy_check_mark:
SLIDER    | 30 | 15  | :heavy_check_mark:
TEXTAREA  | 18 |     | :x:
SPINBOX   | 19 |     | :x:
CPICKER   | 20 |     | :heavy_check_mark:
|**<div align=center>Visualizers</div>**| **id** | **newid** | **implemented** | **note**
LABEL     | 12 | 30 | :heavy_check_mark:
GAUGE     | 31 |    | :heavy_check_mark:
BAR       | 32 |    | :heavy_check_mark:
LMETER    | 33 |    | :heavy_check_mark:
LED       | 41 | 34 | :heavy_check_mark:
ARC       | 22 | 35 | :heavy_check_mark:
PRELOADER | 21 | 36 | :heavy_check_mark:
CHART     | 37 |    | :clock9:
|**<div align=center>Selectors</div>**| **id** | **newid** | **implemented** | **note**
DDLIST    | 50 |     |  :heavy_check_mark:
ROLLER    | 51 |     | :heavy_check_mark: 
LIST      | 52 |     | :x:
TABLE     | 53 |     | :clock9:
CALENDER  | 54 |     | :heavy_check_mark:
|**<div align=center>Containers</div>**| **id** | **newid** | **implemented** | **note**
CONTAINER | 60 |    | :clock9:
WINDOW    | 61 |    | :x:
MSGBOX    | 62 |    | :x:
TILEVIEW  | 63 |    | :x:
TABVIEW   | 64 |    | :x:
TAB       | 65 |    | :x:
PAGE      | 69 |    | :o: | Obsolete in v8
|**<div align=center>Graphics</div>**| **id** | **newid** | **implemented** | **note**
LINE      | 70 |    | :x:
IMAGE     | 71 |    | :x:
CANVAS    | 72 |    | :x:
MASK      | 73 |    | :x:

fvanroie, 2021-01-11:
The change from `objid` to `obj` has been implemented, with `objid` being absoleted, but still working.
The docs have been updated with the new object names which follow the lvgl naming of widgets.

arovak, 2021-01-11:
Looks good, will make managing objects easier for users.

fvanroie, 2021-02-08:
.txt is inconsistent with the rest of the lvgl properties, a .text property would be suited better and in line with the other attributes.
Both can exist while .txt is phased out...

fvanroie, 2021-02-09:
Attribute `text` has been added to replace `txt`, which is now obsoleted...

fvanroie, 2021-04-14:
Using `objid` or `txt` properties now result in an obsolete warning in the logs as a last step to removing them altogether.

nagyrobi, 2021-06-07:
+1 for [spinbox](https://docs.lvgl.io/latest/en/html/widgets/spinbox.html). Especially with regard to supporting decimal values (as text).

closed 2022-05-05 by fvanroie