HASwitchPlate/openHASP #42: roller changed value reports in separate mqtt messages

blakadder, 2021-01-08:
```
x
Jan  8 23:06:50 plate_aad368 MQTT[PUB] -    #033[91m[81008/130612 37]#033[92m[49480/50368  2]hasp/plate_aad368/state/p0b6 => {"val":"1"}                     x
Jan  8 23:06:50 plate_aad368 MQTT[PUB] -    #033[91m[81008/130612 37]#033[92m[49480/50368  2]hasp/plate_aad368/state/p0b6 => {"txt":"Green Tea"}             x
Jan  8 23:06:50 plate_aad368 MQTT[PUB] -    #033[91m[81008/130612 37]#033[92m[49480/50448  2]hasp/plate_aad368/state/p0b6 => {"val":"2"}                     x
Jan  8 23:06:50 plate_aad368 MQTT[PUB] -    #033[91m[81008/130612 37]#033[92m[49480/50448  2]hasp/plate_aad368/state/p0b6 => {"txt":"Oolong Tea"}   
```
roller objects reports value changes in two separate mqtt messages. would probably be better to merge them together

`hasp/plate_aad368/state/p0b6 = {"val":"2","txt":"Oolong Tea"} `

fvanroie, 2021-01-09:
Definitely room for improvement there, this will become one message for roller and dropdown list:

```
[             434.632][81008/128600 37][52924/53616  2] MQTT PUB: hasp/plate35/state/p0b1 => {"val":2,"txt":"Autumn"}
[             438.601][81008/128852 37][52924/53528  2] MQTT PUB: hasp/plate35/state/p0b2 => {"val":1,"txt":"2021"}
```

fvanroie, 2021-01-09:
It should be OK now.

fvanroie, 2021-01-11:
ddlist, roller and btnmatrix all use this new format. List is also a candidate to share this output function.

Table will also follow this syntax but with row,col instead of val.

fvanroie, 2021-02-09:
To be more in line with the lvgl naming convention, the `txt` attribute has been renamed to `text`:
```
[             434.632][81008/128600 37][52924/53616  2] MQTT PUB: hasp/plate35/state/p0b1 => {"val":2,"text":"Autumn"}
[             438.601][81008/128852 37][52924/53528  2] MQTT PUB: hasp/plate35/state/p0b2 => {"val":1,"text":"2021"}
```

closed 2021-02-09 by fvanroie