HASwitchPlate/openHASP #43: dropdown selection does not respect text_font size

blakadder, 2021-01-08:
[image]

object setting `{"objid":50,"id":6,"x":55,"y":150,"w":150,"options":"Coffee\nGreen Tea\nOolong Tea\nPu-Erh Tea\n","text_font":12}`

default font is set to very large so it renders other options using it instead of text_font

fvanroie, 2021-01-17:
I think this is related to the custom HASP theme. Does it happen with Material too?

blakadder, 2021-01-17:
That was the material theme used, its my default goto

fvanroie, 2021-01-17:
OK I'll look into this a bit further

htvekov, 2021-02-05:
I'm experiencing same issue using default font
hasp Dark theme (hasp v0.3.3 - ESP32)

object:`{"obj":"dropdown","id":25,"page":4,"x":1,"y":1,"w":239,"h":319,"bg_color":"#000000","bg_opa":255,"border_color":"#C7BAA7","border_width":1,"val":2,"options":"TEST1\nTEST2\n","txt":"SOURCE","text_color":"#FFAC00","text_font":22}`

[image]

fvanroie, 2021-02-09:
This issue should now be fixed.

closed 2021-02-11 by fvanroie