HASwitchPlate/openHASP #47: memory leak after running for a while?

blakadder, 2021-01-17:
```
x
Jan 17 04:52:24 lanbon ATTR    #033[91m[81008/127792 36]#033[91m[  644/  776 18].value_str (1091)                                                                                                                         x
Jan 17 04:52:24 lanbon LVGL    #033[91m[81008/127872 36]#033[91m[  200/  544 64]Couldn't allocate memory                                                                                                                  x
Jan 17 04:52:24 lanbon LVGL    #033[91m[81008/127872 36]#033[91m[  200/  544 64]Couldn't allocate memory                                                                                                                  x
Jan 17 04:52:24 lanbon LVGL    #033[91m[81008/127872 36]#033[91m[  200/  544 64]_lv_mem_buf_get                                                                                                                           x
Jan 17 04:52:24 lanbon LVGL    #033[91m[81008/127872 36]#033[91m[  200/  544 64]Out of memory, can't allocate a new  buffer (increase your LV_MEM_SIZE/heap size (0x00000000)   
```

fvanroie, 2021-01-17:
Commit 647d1bc1f2d8a48601d804563cbd7607f5bb740a fixed a memory leak with value_str. I think that might be the cause of this issue. Could you a the latest version and see if that fixes it?

blakadder, 2021-01-17:
I'm leaving it running now so we'll see

closed 2021-02-01 by fvanroie