HASwitchPlate/openHASP #49: Page command enhancement

blakadder, 2021-01-28:
Add a `page next/previous` or `page +/-` commands that would switch to next configured page then loop back to page 0 once it reaches the end number of configured pages. Useful for page 254 buttons to go to next/previous page

add extra resilience for page command so it doesn't default to 0 when using anything but numbers but does nothing and returns error/invalid command in log

```
Jan 28 19:13:07 hasptest MQTT[RCV] - #033[91m[81008/136116 40]#033[92m[54588/54612  1]hasp/hasptest/command/page = next
Jan 28 19:13:07 hasptest HASP #033[91m[81008/136116 40]#033[92m[54588/54612  1]Changing page to 0
Jan 28 19:13:07 hasptest HASP #033[91m[81008/136116 40]#033[92m[54588/54612  1][0] p[0].b[0] lv_obj
Jan 28 19:13:07 hasptest MQTT[PUB] - #033[91m[81008/132944 39]#033[92m[54588/54612  1]hasp/hasptest/state/page => 0
Jan 28 19:13:16 hasptest MQTT[RCV] - #033[91m[81008/136116 40]#033[92m[54588/54612  1]hasp/hasptest/command/page = +
Jan 28 19:13:16 hasptest HASP #033[91m[81008/136116 40]#033[92m[54588/54612  1]Changing page to 0
```

fvanroie, 2021-01-29:
Page 254 has been renumbered to page 0. With 1-12 as 'normal' pages.

page `prev`/`next` has been added:
```
[Jan 29 16:14:52.541][81008/128312 36][56808/56960  1] MSGR: page = next
[Jan 29 16:14:52.544][81008/128312 36][56808/56960  1] HASP: Changing page to 2
[Jan 29 16:14:52.557][81008/128312 36][56808/56960  1] HASP: [0] p[2].b[0] lv_obj
[Jan 29 16:14:52.574][81008/125048 35][56808/56960  1] MQTT PUB: hasp/plate35/state/page => 2
[Jan 29 16:14:57.588][81008/128312 36][56808/56960  1] MSGR: page = prev
[Jan 29 16:14:57.591][81008/128312 36][56808/56960  1] HASP: Changing page to 1
[Jan 29 16:15:05.294][81008/125276 35][56808/56960  1] MQTT PUB: hasp/plate35/state/page => 1
```
It just goes through all the pages at the moment, as there is no notion of *configured pages* yet.

Invalid pages:
```
[Jan 29 16:15:05.276][81008/128312 36][56808/56960  1] MSGR: page = 113
[Jan 29 16:15:05.278][81008/128312 36][56808/56960  1] HASP: Invalid page 113
[Jan 29 16:15:12.851][81008/128312 36][56808/56960  1] MSGR: page = abc
[Jan 29 16:15:12.853][81008/128312 36][56808/56960  1] MSGR: Invalid page abc
```

closed 2021-01-29 by blakadder

fvanroie, 2021-01-29:
Docs updated