HASwitchPlate/openHASP #51: Enhancement: Adding optional entity_id and yaml/json key to each object for easier Home Assistant templating

htvekov, 2021-01-30:
Hi' Francis.

WOW !! What an amazing piece of software!! 👏👏😎
It's almost unbelieable that you've managed to squeeze all this into an ESP8266 🤯

I've had a Nextion display lying around for almost a year, but never really got to setup and test good old HASP.
And now I've stumpled across this little gem by pure accident - lucky me 😉

Hasp-lvgl can do all I searched for and a lot more.
I'm using Home Assistant to tie up all my IoT and automations.

Could you add the possibility to specify and tie an optional HA entity_id and key with an object in the pages.jsonl file ?
That would make templating in various automations possible for groups of 'identical' entities (light, switches etc.)

The easy part would be to include these in the state json attributes string thas is published.
But HA would need to push commands and state changes as well to hasp-lvgl.
As page and object number is arbitrary to HA, entity_id would have to be used as identifier in topic, 

Can't really think of any easier and more clever method to link entity_id together with page and object number 🤔

Have I completely missed an obvious solution that could be used as alternative ? 😆

Rgds.
Henning

blakadder, 2021-01-30:
All that can be accomplished with blueprints/ automations which are currently under development.

Some very alpha WIP can be found here https://github.com/blakadder/hasplvgl-hablueprints

fvanroie, 2021-01-30:
Hello Henning,

I am a HA n00b myself, so I don't fully have a complete view of how all the pieces fit together *yet*. 😃

We have some very competent people that are working on the integration of HA and hasp-lvgl, like @blakadder.
Most of this discussion happens in the [#hasp-lvgl discord channel](https://discord.gg/VCWyuhF), feel free to join us there for up-to-date HA changes.

Between auto-discovery, blueprints and automations the goal is to make it user-friendly and easy to use in HA.
It is starting to come together,  with improvements made almost every day.

Check out the blueprints and let us know if that answers your question.

htvekov, 2021-01-31:
Hi' @fvanroie & @blakadder 

Just had a quick look at the discord channel and it pretty much looks like you on fire - the bunch of you 🔥😉
I sense that code development is rapid - more or less 24/7 🚀🚀

Well, some 24 hours after initial start-up I'm still impressed and can see that as a complete amateur 'tinkerer', I'll have a steep learning curve ahead of me.

I'll checkout the blueprints  for sure. Newer used these before, though.

I'm 'brought up' with yaml since my HA startup some 2 years ago.
Though I'll still find myself sometimes spending hours looking non existing errors, only to find one stupid typo or an extra whitespace somewhere in the yaml or templates 🙄 I'm to old for that kind of syntax ! In fact so old, that I spend most of my teenage years doing z80 machine code. Talk about non existing syntax check ! 😆

I'll use discord for future discussions and minor issues  👍

Ciao !

closed 2021-01-31 by htvekov