HASwitchPlate/openHASP #55: lanbon l8 initial setup challenges

jinhong-, 2021-02-06:
Version tested on: v0.3.2, v0.3.1
Device: lanbon l8

I want to start by saying thanks for this wonderful project. It was exactly what i have been looking for! I recently purchased a lanbon l8 to give it a go. I just wanted to share some of the initial challenges i had faced:

1. After flashing the device, i successfully connected to the temp wifi network for doing configuration. However, accessing http://192.168.4.1 did not load for me. I have tried this on both iOS and Android with no success. Everything else was working fine. It looks like DHCP assigned my phone an IP of 192.168.4.2
2. Then i went to try entering my network information manually. This however did not work as well. On pressing okay, it would hang for a few seconds and do nothing
3. I finally proceeded to manually compile a version myself, configuring the information directly using `user_config_override.h`. It then complained that my wifi password was too long, which i proceeded to change this line `char wifiPassword[32] = WIFI_PASSW;` to match my password length in `hasp_wifi.cpp`

I hope this will help other people who are also trying the lanbon l8

fvanroie, 2021-02-07:
Thanks for your report of the setup experience. I will need to have a closer look at these steps again, as it is meant to be as seamless as possible. Good to hear you finally managed to connect the device.

I will leave this issue open until I can investigate the root cause of this issue.

nagyrobi, 2021-02-10:
Hi fvanroie please do have a look as my Lanbon L8 is on the way to me too and I bought it only for the reason of being supported by your firmware.

yvesdm3000, 2021-02-19:
I had the same issue, I could connect to the AP but no webserver was running on 192.168.4.1. Entering the network settings on-screen worked fine.

BansEN77, 2021-02-21:
I confirm this happened as well for me.

fvanroie, 2021-02-22:
OK, we think to have found the culprit and I put a fix in the [nightly build](https://github.com/fvanroie/hasp-lvgl/actions/runs/590771228) of the master branch if anyone cares to test it.

If it is confirmed I'll make 0.3.4 release.

fvanroie, 2021-02-23:
v0.3.4 has been released with a fix for this issue. Feel free to try it out and let me know if you still experience this issue.

dgomes, 2021-02-24:
sorry to hijack the thread, but at least @jinhong- has already flashed his lanbon 😄 

- what pin is GPI0 in lanbon breakout connector ? Or do we need to solder the pin directly ?

fvanroie, 2021-02-24:
> sorry to hijack the thread, but at least @jinhong- has already flashed his lanbon 😄

You can use the Q&A section in the [Github discussions](https://github.com/fvanroie/hasp-lvgl/discussions/categories/q-a) forum for getting help. We are also on [Discord](https://discord.gg/VCWyuhF) if you prefer that.
 
>     * what pin is GPI0 in lanbon breakout connector ? Or do we need to solder the pin directly ?
You can use the pin marked `IO0` on the pinheader. There is no RESET pin broken out, so you need to power-cycle the board to get into flash mode. If all else fails, you can also [solder the chip directly](https://blakadder.com/lanbon-L8-custom-firmware/).

yvesdm3000, 2021-02-24:
I just connected some breadboard wires into the lanbon connector and connected to my CH340 usb ttl esp-01 flasher tool. The pins are nicely marked so it was a breeze to flash the Lanbon L8.

fvanroie, 2021-02-25:
This issue has now been resolved. We have opened a [discussion topic](https://github.com/fvanroie/hasp-lvgl/discussions/65) for the Lanbon L8 to exchange experiences.

closed 2021-02-25 by fvanroie

spp12345, 2021-04-23:
i have problem.
when i flash lanbon l8 via esp flasher i get an error:

`Using 'COM10' as serial port.
Connecting........_____....._____....._____....._____....._____....._____....._____
Unexpected error: ESP Chip Auto-Detection failed: Failed to connect to Espressif device: Timed out waiting for packet header
`
what can i do ?

fvanroie, 2021-04-23:
Try any of the tools mentioned in [this post](https://github.com/HASwitchPlate/openHASP/issues/143#issuecomment-822136123).
ESPHome-Flasher doesn't seem to work as well as the others.

spp12345, 2021-04-23:
when i using Tasmota-PyFlasher-1.0
`Command: esptool.py --port COM4 --baud 115200 --after no_reset write_flash --flash_mode dout 0x00000 C:\Users\xxx\Desktop\lanbon_l8_full_8MB_v0.5.0.bin --erase-all

esptool.py v2.8
Serial port COM4
Connecting........_____....._____....._____....._____....._____....._____....._____`

and when flash_download_tool_v3.8.5 i have the same :( _____....._____....._____....._____....._____....._____....._____

what do you think what can i do ?

fvanroie, 2021-04-23:
It seems that the ESP32 is not booted into flash mode:
- Make sure you connect pin `IO0` to `GND`
- Completely power off the device
- Power on the device
- Try to upload again

Also double-check that you are using the correct COM-port to which the device is connected.

spp12345, 2021-04-23:
ok!!! thank you very much!! i swapping RX-TX pins and works great !!!!!