HASwitchPlate/openHASP #56: [Question] Loading of fonts

jinhong-, 2021-02-08:
Version tested on: v0.3.2
Device: lanbon l8

I recently tried to load fontawesome into my device but I have had very little success in getting it to work. Do I also need custom fonts in order for fontawesome to load?

fvanroie, 2021-02-11:
Yes, the fontawesome font is an extension of another .zi font of the same size.

jinhong-, 2021-02-11:
Thanks
.zi fonts are hard to find. Might be a silly question but any chance that standard fonts would be supported? For eg, OTF fonts

fvanroie, 2021-02-12:
OTF fonts are vector based and would take a lot of resources to render on the microcontroller.
.zi is the font format used in Nextion/TJC projects. They can be created using their editors.

We created the [HMI Font Pack](https://github.com/fvanroie/HMI-Font-Pack) with a lot of pre-created fonts. Only ascii/latin-1 encoding is currently supported in hasp-lvgl.
.zi Fonts are currently still buggy and might crash... better font support is planned.

jinhong-, 2021-02-12:
Thanks @fvanroie. It looks like lanbon L8 does not really have that much memory to work with. The font packs comes with fonts that are quite large. I learnt that I could use the [offline font converter](https://github.com/lvgl/lv_font_conv) or the [online one](https://lvgl.io/tools/fontconverter) to embed more fontawesome icons into the built in fonts. That's probably what i am going to do

closed 2021-02-12 by jinhong-

nagyrobi, 2021-03-26:
See: https://github.com/nagyrobi/files/tree/main/MaterialDesignIcons-4-hasp-lvgl

fvanroie, 2021-03-26:
While that approach will certainly work, it also uses about 50Kb of additional memory or more in flash _and_ sram. You need to pick and choose which icons to included as memory is very limited.

The whole point of .zi fonts is to be able to upload custom fonts that are not constantly loaded in sram, but are displayed on-the-fly when needed.

nagyrobi, 2021-03-26:
That's a great idea but at the moment the possibilities are somewhat limited by the fact that I lack the knowledge and tools to create .zi fonts, plus my recent experience was that .zi support is heavily still in progress, since after uploading and setting a .zi as default I ended up with a brick which needed a reflash...

fvanroie, 2021-03-27:
in the latest master the .zi fonts should work again and the crash bug was found and fixed.

RossiRPI, 2021-04-08:
@fvanroie do you have a guide on how to create custom .zi font packs? I see that you have been involved in the project but I cant figure out how to get the icons from font awesome and create my own pack.
The current font sizes implemented in the ESP32 releases is a bit too small for the wt21-sc01's 320x480 display.

fvanroie, 2021-04-10:
There is no guide to create .zi fonts, only the file format description and Nextion Font Editor tools. These are open source, so you can check them out to see what they do.

You can just download the [HMI Font Pack](https://github.com/fvanroie/HMI-Font-Pack/releases/tag/v0.0.2-alpha) as these contain hundreds of pre-built .zi fonts, including fontawesome.