HASwitchPlate/openHASP #59: L8 Moodlight

mssaleh, 2021-02-15:
First of all, Thanks for the massive, impressive effort. I have been trying to make use of these high-potential devices for more than a year. I managed to create several working prototypes using vanilla LVGL but no 2-way integration with HomeAssistant, and also with ESPhome (works nicely with HA but no touch!). I am beyond grateful for your generous contribution here.

Regarding the L8. Now with version 0.3.3, when I set up the backlight of L8 using moodlight  (GPIOs r:26,g:32,b:33). The resulting entity in HomeAssistant (via MQTT auto-discovery) isn't working. I tried to create a manual MQTT light using the MQTT Light Template Schema in HA which worked only for on/off but not for color.
I created 3 sliders on a page and used groupid to link them to each GPIO directly from the HASP without HomeAssistant. This seems to work fine. Kindly take a look if have an L8.

Also, in general, I think it is possible to link a color-picker object with moodlight using groupid so we can directly control it without MQTT. It needs a bit of work but will make more sense than using HA or MQTT to change the backlight color on L8.

Best Regards!

mssaleh, 2021-02-16:
I tried to read the code parts related to moodlight and groupid in order to help and contribute to this enhancement, but because I don't have enough C++ knowledge and there is very little comments in the code, I haven't got the confidence to submit a pull request.

My simple idea with minimal code change is to assign a pre-fixed groupid to moodlight (such as 888) and then modify to value normalization function to allow for the color hex code (with a simple "if" statement to create exception for groupid 888).

fvanroie, 2021-02-16:
Hi @mssaleh,

Thanks for the input, I'm looking into this... The moodlight uses 3 groupids, one per PWM channel.
The way groups are currently setup, each object belongs to a single groupid. A colorpicker is special in that regard...

I think for HA to recognize the RGB values, the message needs to be in {"rgb":[255,255,255]}format.
I'm a HA n00b so I need to test that part some more, but this is certainly a goal to have that working too.

fvanroie, 2021-02-23:
OK, I got the moodlight auto-discovery working in HA now.

mssaleh, 2021-02-24:
Thanks a lot! You are really a github treasure! I have been extremely busy at work but I am planning to contribute to your effort in March.

fvanroie, 2021-02-27:
[image]

Looking good, still need to add brightness support.

nagyrobi, 2021-03-04:
How do you guys use this with autodiscovery in HA?
I configure everything manually - can this be done too?
Any HA config sample perhaps?

fvanroie, 2021-03-05:
autodiscovery is not yet working in 0.3.4, but in 0.4.0-dev it should work now

nagyrobi, 2021-03-05:
But is there a way to use this with manual confuguration, not autodiscovery?

fvanroie, 2021-03-05:
There is a [moodlight command](https://fvanroie.github.io/hasp-docs/#commands/#moodlight) in 0.3.4

nagyrobi, 2021-03-08:
Ok my problem was that my Home Assistant's MQTT was initially configured back when autodiscovery didn't even exist. Somehow autodiscovery refused to work but re-installing the MQTT integration solved this.

nagyrobi, 2021-03-29:
> But is there a way to use this with manual confuguration, not autodiscovery?

The custom component adds it manually so no more autodiscovery needed, cool.

fvanroie, 2021-03-29:
autodiscovery was the only possibility for a while, this has been dropped now since we have a custom component that can handle it better.

nagyrobi, 2021-03-29:
yes, that's why I thought I should mention this.

fvanroie, 2021-04-14:
I think we can close this issue in favor of the [openHASP custom component](https://github.com/HASwitchPlate/openHASP-custom-component)?

closed 2021-04-14 by mssaleh

nagyrobi, 2021-04-14:
> Looking good, still need to add brightness support.

Is brightness of moodlight implemented at firmware level? Or is it just missing from the component?