HASwitchPlate/openHASP #72: Lanbon L8 build error using platformio-core

nagyrobi, 2021-03-04:
### Perform all steps below and tick them with [x]
- [x] Check the related part of the [Documentation](https://fvanroie.github.io/hasp-docs/)
- [x] Update hasp-lvgl to the latest version
- [x] Reproduce the issue and describe all steps

### Describe the bug

Build for Lanbon L8 fails with:
`Error: The program size (1447642 bytes) is greater than maximum allowed (1441792 bytes)`
Nevertheless tried to flash to the unit, that didn't produce an error. But the device doesn't boot, only a white screen appears.
### To Reproduce

- Clone repo
- Create `platformio_override.ini` where I only uncomment `user_setups/esp32/*.ini` and `lanbon_l8`
- Build with `pio run 2>&1 | tee build.log`

See: [build.log](https://github.com/fvanroie/hasp-lvgl/files/6084684/build.log)
### Expected behavior

Have a build without errors.
### Screenshots or video

fvanroie, 2021-03-04:
The recent espressif v3.1.0 is quite a bit larger then v3.0.0, causing the build to fail.
I pinned it to the previous `platform = espressif32@3.0.0` for Lanbon L8 builds now.

nagyrobi, 2021-03-04:
OK testing...

nagyrobi, 2021-03-04:
Still not good: [build.log](https://github.com/fvanroie/hasp-lvgl/files/6086050/build.log)

fvanroie, 2021-03-04:
Which branch are you building?

nagyrobi, 2021-03-04:
The default one which comes with `git clone --recursive https://github.com/fvanroie/hasp-lvgl` as stated in the main [readme](https://github.com/fvanroie/hasp-lvgl/blob/master/README.md).

fvanroie, 2021-03-04:
OK I need to investigate this a bit further. In the meantime, you can use the auto-compiled binary from https://github.com/fvanroie/hasp-lvgl/actions/runs/622049076. The linux build seems to build find, while on Windows it fails...

nagyrobi, 2021-03-04:
I'm on Linux too..

nagyrobi, 2021-03-05:
Maybe because BLE stack included with Lanbon L8?

```
Compiling .pio/build/lanbon_l8/lib079/BLE/BLE2902.cpp.o
Compiling .pio/build/lanbon_l8/lib079/BLE/BLE2904.cpp.o
Archiving .pio/build/lanbon_l8/libb0f/libAceButton.a
Compiling .pio/build/lanbon_l8/lib079/BLE/BLEAddress.cpp.o
Indexing .pio/build/lanbon_l8/libb0f/libAceButton.a
Compiling .pio/build/lanbon_l8/lib079/BLE/BLEAdvertisedDevice.cpp.o
Compiling .pio/build/lanbon_l8/lib079/BLE/BLEAdvertising.cpp.o
Compiling .pio/build/lanbon_l8/lib079/BLE/BLEBeacon.cpp.o
Compiling .pio/build/lanbon_l8/lib079/BLE/BLECharacteristic.cpp.o
Compiling .pio/build/lanbon_l8/lib079/BLE/BLECharacteristicMap.cpp.o
Compiling .pio/build/lanbon_l8/lib079/BLE/BLEClient.cpp.o
Compiling .pio/build/lanbon_l8/lib079/BLE/BLEDescriptor.cpp.o
Compiling .pio/build/lanbon_l8/lib079/BLE/BLEDescriptorMap.cpp.o
Compiling .pio/build/lanbon_l8/lib079/BLE/BLEDevice.cpp.o
Compiling .pio/build/lanbon_l8/lib079/BLE/BLEEddystoneTLM.cpp.o
Compiling .pio/build/lanbon_l8/lib079/BLE/BLEEddystoneURL.cpp.o
Compiling .pio/build/lanbon_l8/lib079/BLE/BLEExceptions.cpp.o
Compiling .pio/build/lanbon_l8/lib079/BLE/BLEHIDDevice.cpp.o
Archiving .pio/build/lanbon_l8/lib083/libConsoleInput.a
Archiving .pio/build/lanbon_l8/libda1/libEEPROM.a
Indexing .pio/build/lanbon_l8/lib083/libConsoleInput.a
Indexing .pio/build/lanbon_l8/libda1/libEEPROM.a
Compiling .pio/build/lanbon_l8/lib079/BLE/BLERemoteCharacteristic.cpp.o
Compiling .pio/build/lanbon_l8/lib079/BLE/BLERemoteDescriptor.cpp.o
Compiling .pio/build/lanbon_l8/lib079/BLE/BLERemoteService.cpp.o
Compiling .pio/build/lanbon_l8/lib079/BLE/BLEScan.cpp.o
Compiling .pio/build/lanbon_l8/lib079/BLE/BLESecurity.cpp.o
Compiling .pio/build/lanbon_l8/lib079/BLE/BLEServer.cpp.o
Compiling .pio/build/lanbon_l8/lib079/BLE/BLEService.cpp.o
Compiling .pio/build/lanbon_l8/lib079/BLE/BLEServiceMap.cpp.o
Compiling .pio/build/lanbon_l8/lib079/BLE/BLEUUID.cpp.o
Archiving .pio/build/lanbon_l8/lib06a/libPubSubClient.a
Indexing .pio/build/lanbon_l8/lib06a/libPubSubClient.a
Compiling .pio/build/lanbon_l8/lib079/BLE/BLEUtils.cpp.o
Compiling .pio/build/lanbon_l8/lib079/BLE/BLEValue.cpp.o
Compiling .pio/build/lanbon_l8/lib079/BLE/FreeRTOS.cpp.o
Compiling .pio/build/lanbon_l8/lib079/BLE/GeneralUtils.cpp.o
```
Afaik this device doesn't support Bluetooth so maybe this just takes up the space?

nagyrobi, 2021-03-05:
I succeeded to create an image which fits in by removing `robotocondensed_regular_28_nokern.c`.
Since it [doesn't seem](https://github.com/fvanroie/hasp-lvgl/discussions/74) support LATIN-2 or Unicode chars, it needs replacement anyway (at least for me)

nagyrobi, 2021-03-08:
0.4.0-dev now builds like this:

Visual Studio + Platformio on a Linux desktop environment:
`Flash: [==========]  97.9% (used 1411030 bytes from 1441792 bytes)`

Platformio-Core only on a linux VM envoronmnt:
`Flash: [==========]  100.5% (used 1448530 bytes from 1441792 bytes)`

Really strange...

fvanroie, 2021-03-18:
The partition size for 0.4.0-dev is now increased to 2Mb.

nagyrobi, 2021-03-18:
Great! What's the flash size of Lanbon L8 btw?

fvanroie, 2021-03-18:
It's 8Mb

closed 2021-03-18 by nagyrobi