HASwitchPlate/openHASP #80: Feature request: with screen off, first touch should just wake screen

nagyrobi, 2021-03-06:
After Long Idle, screen turns off. If I touch the screen to wake it, the object underneath will also react. This is a problem as the user might touch in the wrong area without first seeing the screen, causing undesired effects. 

It would be nice to have an option/setting which enabled, at touch would check first if the screen is off and if it is, just wake it and exit routine/function without touching the objects.

fvanroie, 2021-03-08:
This request has come up before and will be included in one way or another...

fvanroie, 2021-03-09:
This is now implemented in the 0.4.0-dev branch.

nagyrobi, 2021-03-10:
Works, excellent, thank you!

nagyrobi, 2021-03-14:
No it doesn't... so reopening...

fvanroie, 2021-03-16:
Are you setting dim to 0 instead of setting light to off?

The wakeup touch enables the backlight, so reverting to the previous dim level.
If the dim level is 0, the backlight will turn on to the level 0...

nagyrobi, 2021-03-16:
I'm using thh HA component...

nagyrobi, 2021-03-16:
No that's not the problem. The problem is that if I touch the screen it wakes up normally, but it also clicks the object.

fvanroie, 2021-03-16:
Wakeup touch is only enabled/disabled when using `light off/on` commands. You should see it appear in the log:

[image]

It seems you are just setting dim to 0, which will not enable wakeup touch.

fvanroie, 2021-03-26:
Is this still an issue?

nagyrobi, 2021-03-27:
Hopefully not. I expeience strange touch things occasionally but i think not related to this.

closed 2021-03-27 by nagyrobi