HASwitchPlate/openHASP #81: Feature request: decimal values for slider and arc (adjustable)

nagyrobi, 2021-03-07:
Currently `slider` (+double slider) and `arc` (adjustable) only support integer values (`int16`).
Would be useful (for a [thermostat control](https://github.com/fvanroie/hasp-lvgl/discussions/37#discussioncomment-433349) for example) to use decimals too. Perhaps `float16` (if supported) would be enough for this. A new property `precision` would be required to set precision:
- `0` (default) for no decimal (would behave just like now)
- `1` for one decimal `.1` -> `.9`
- `2` for two decimals `.01` -> `.99`

fvanroie, 2021-03-07:
Lvgl has only integer values for object values. Trying to use floats and converting on-the-fly on the MCU doesn't seem the way to go here. This can also be solved by just multiplying your min/max values by the precision (*10 or *100) on the automation system.

Then devide the value reported by the same base to get the float value.

nagyrobi, 2021-03-07:
Ok, sent a note to https://github.com/dgomes/hasp-lvgl-custom-component/issues/9

fvanroie, 2021-03-07:
I'll close this issue and discuss with @dgomes. We can revisit this feature when time comes.

closed 2021-03-07 by fvanroie