HASwitchPlate/openHASP #99: Group items state after reboot / powercycle

nagyrobi, 2021-03-18:
Group items state is lost after reboot. To reproduce:
- Put a toggle button or switch object, and a relay in the same group.
- touch the object, it's value will be `on` the relay turns `on`
- reboot the plate (from the web interface)
- during boot the relay stays `on` (correctly), but after boot the object displays as `off`, state in MQTT is `on` (as no update came)

Also inconsistent state occurs after power-cycling. To reproduce:
- Put a toggle button or switch object, and a relay in the same group.
- touch the object, it's value will be `on` the relay turns `on`
- power-cycle the plate. The relay turns off due to power loss
- during boot the relay stays `off` (correctly), after boot the object displays as `off`, state in MQTT is `on`, incorrect (as no update came)

I don't think we should save the state to the flash, instead we should check the state of the GPIO outputs in the group and set the objects state the objects accordingly. Also send an MQTT about this to make sure real state is known after reboot or power cycling.

nagyrobi, 2021-03-18:
Please note that Github labels the issues automatically as bug and assigns them to @fvanroie , it was not me who puts these labels on. I don't have permission to change these afterwards.

fvanroie, 2021-03-18:
> 
> 
> Please note that Github labels the issues automatically as bug and assigns them to @fvanroie , it was not me who puts these labels on. I don't have permission to change these afterwards.

When you open a bug report, these get added automatically. Opening a Feature request or question will add the appropriate labels too. When in doubt, join our discussion forum or Discord channel and we'll happily discuss it there first.

fvanroie, 2021-05-31:
Moved from #123 

**Is your feature request related to a problem? Please describe.**
Similar to [group items](https://github.com/HASwitchPlate/openHASP/issues/99), moodlight color, state (and [brightness](https://github.com/HASwitchPlate/openHASP/issues/59#issuecomment-787106735) after implemented) should be remembered after reboot / powercycle.
If HA and plate both reboot _while the moodlight was turned off_, nobody remembers what was the pre-selected moodlight attribute. At HA reboot, the current of moodlight color is read from the plate because HA [doesn't keep the color attribute](https://github.com/home-assistant/core/issues/2316) when light is off. If plate also reboots during this time (eg. power outage affecting all devices), nobody remembers what was the pre-selected color.

**Describe the solution you'd like**
When `moodlight` command comes in with `color` key, check if moodlight color info is saved to the flash. If moodlight color info in flash is different of what came with the command, write to flash the new color info (save flash writes if every `{"state":"true"}` command comes with the same color info every time...)
At boot restore moodlight color info (and state) from flash.

**Describe alternatives you've considered**
Only this

**Additional context**
Not applicable

fvanroie, 2021-08-30:
Atm, joining an object or gpio to a group will not set the item to the last known group state.
In fact, the group state is not kept at all but only fired on change events.

So this would require keeping the last group state, on join: set new members of the group to that state and advertise the new value of the item.

closed 2024-02-17 by nagyrobi