HASwitchPlate/openHASP #114: number of pages

nagyrobi, 2021-03-29:
A quick note: `numPages` in `statusupdate` is always 12, while I only defined 5 pages in my .jsonl file.

fvanroie, 2021-03-29:
yes

nagyrobi, 2021-03-29:
I thought `numPages` would reflect the number of configured pages (aka. pages with at least 1 object on them).
Would be useful to prevent switching to empty pages when keep pressing next.

fvanroie, 2021-03-30:
`numPages` reflects the maximum number of pages (aka. the maximum number of the last page).

hasp-lvgl has no concept of an *empty* page. All pages exist at all times and are available to create/delete objects whether you use them or not.

nagyrobi, 2021-03-30:
It would be nice to differentiate somehow the pages containing meaningful objects, so that the end-user should not be presented with a page containing irrelevant objects or no objects at all (=empty).

I thought this could be known automatically just by looking at the "page" property of all the objects, and take the highest number from there. That's all.

fvanroie, 2021-04-17:
You can now get/set the following attributes on a page object (e.g. `p1b0`):
- `prev` : The number of the destination page when performing a `page prev` action on this page
- `next` : The number of the destination page when performing a `page next` action on this page
- `back` : The number of the destination page when performing a `page back` action on this page

By default all pages cycle in a round-robbing fashion, but you can change this behaviour by adjusting the `prev`, `next` and `back` target pages:
```json
p1b0.prev=5    // go to page 5 instead of page 12
p1b5.next=1    // start back from page 1
```

This will cycle through page 1-5 only. `back` acts like a * level up* action, so you can jump back to the home or menu page where you came from using `page back`.

nagyrobi, 2021-04-17:
How can this be set from jsonl?

nagyrobi, 2021-04-17:
@dgomes  also CC support has to be added as currently the component calls pages directly by number when next/prev

dgomes, 2021-04-17:
if the firmware handles this I can remove page control altogether :)

fvanroie, 2021-04-17:
> How can this be set from jsonl?

`jsonl`

```json
{"page":1,"id":0,"next":1,"prev":5,"back":3,"bg_color":"black"}
```

fvanroie, 2021-04-21:
This was implemented in 0.5.0

nagyrobi, 2021-04-21:
Works, thanks

closed 2021-04-21 by nagyrobi