HASwitchPlate/openHASP #119: Reduce colorpickers value updates

fvanroie, 2021-04-02:
**Is your feature request related to a problem? Please describe.**
haswitchplate/openHASP-custom-component#36

**Describe the solution you'd like**
Update event handler like we did with sliders

**Describe alternatives you've considered**
n/a

**Additional context**
[video](https://user-images.githubusercontent.com/1550668/113125259-b6476480-9216-11eb-8b8e-2480bb51f9f2.mp4)

nagyrobi, 2021-04-02:
Actually, we need this for all objects sending "changed". dropdown, slider, etc.

fvanroie, 2021-04-14:
For normal objects and buttons, I've renamed "short" and "up" events to "up" and "release":
- `up` indicates a short click
- `release` indicates the plate is released after `long` or `hold`events

Both `up` or `release` states indicate the plate is no longer touched. While `down` is sent the object is touched.

fvanroie, 2021-04-14:
Toggle, switch and checkbox have also been updated with an "up" and "down" `event` plus the `val` field.

fvanroie, 2021-04-21:
This was implemented in 0.5.0

closed 2021-04-21 by fvanroie