HASwitchPlate/openHASP #121: Orientation on WT32-SC01 Doesn't Rotate Touch

andersonsmith, 2021-04-03:
First off, I came across this today as I was looking for a way to turn a WT32-SC01 into a thermostat to integrate into homeassistant.  This project is fantastic, thank you for it.

### Perform all steps below and tick them with [x]
- [X ] Check the related part of the [Documentation](https://haswitchplate.github.io/openHASP-docs/)
- [ X] Update openHASP to the latest version
- [X ] Reproduce the issue and describe all steps

### Describe the bug

Setting the orientation to anything other than 0 rotates the display but the touch is still detected in the 0 degree orientation.

### To Reproduce

Confirmed to be consistent at 90, 180, and 270 degree rotation.

### Expected behavior

Touch follows rotation setting the same as display.

### Screenshots or video

fvanroie, 2021-04-03:
At the moment, only the XPT2046 resistive controller is handled automatically by the firmware.
For other touchscreens, there's an lvgl feature I'd like to use but was still a little buggy lvgl/lvgl#2046.
I still need to test and incorporate that fix into the touch driver for those screens.

As a workaround, you can build the firmware from source code and manually set the rotation parameters for the touchscreen.

andersonsmith, 2021-04-03:
Thanks for the reply.  I was able to manually change the touch rotation using `hasp_drv_touch.cpp` by borrowing the code you have commented out.  Specifically, I was looking to use a rotation of 90 for my case and I made the change to the following:

```
    int16_t inv_x;

    inv_x = TFT_WIDTH - normal_x;

    *touchX = normal_y;
    *touchY = inv_x;   
```

fvanroie, 2021-05-31:
This has been fixed/added in 0.6.0 master branch. Please test it and let me know if there are any issues.

closed 2021-06-01 by fvanroie