HASwitchPlate/openHASP #122: cpicker not always accepting color value

nagyrobi, 2021-04-03:
### Perform all steps below and tick them with [x]
- [x] Check the related part of the [Documentation](https://haswitchplate.github.io/openHASP-docs/)
- [x] Update openHASP to the latest version
- [x] Reproduce the issue and describe all steps

### Describe the bug
cpicker object doesn't always accept the color value

### To Reproduce
Using v0.4.1.
This [example](https://haswitchplate.github.io/openHASP-docs/#custom-component/sampl_conf/#rgb-light-color) on a Lanbon L8.
Watch MQTT messages with mosquitto_sub.
In HA you are able to turn on backlight color of the plate. 
On plate possible to select color and it gets updated accordingly in HA.
But in HA if you start clicking on colors (one color per second), the cpicker doesn't always sync to the color received in MQTT message.
Messages like `hasp/plate/command/p7b3.color #ff07e` come correctly each time, but in 20% of the cases cpicker doesn't react to this. Moodlight changes every time correctly.

### Expected behavior
Every time I select in color dial of HA, the MQTT message appears correctly instructing cpicker to change color, and cpicker should do this immediately.

### Screenshots or video
not applicable

fvanroie, 2021-04-24:
Without actual logs, it is hard to figure out what is going on and why messages are being rejected...
I have been running automated test with large numbers of messages at a fast pace.

`hasp/plate/command/p7b3.color #ff07e` actually is invalid because there need to be 3 or 6 hexadecimal numbers following the hash.

Could you open a telnet session to the plate while sending these messages and see if there are any warnings or patterns that might explain more?

nagyrobi, 2021-04-24:
Ok

closed 2021-04-24 by nagyrobi

fvanroie, 2021-04-24:
Was this resolved?

nagyrobi, 2021-04-25:
Couldnt reproduce on late builds. I'll reopen if I notice again.