HASwitchPlate/openHASP #134: Feature request: screen burn-in protection

nagyrobi, 2021-04-09:
**Is your feature request related to a problem? Please describe.**
To protect and prolonge the lifetime of the devices, thus being more green and generating less hazardous waste.
LCD switche's main problem is that they display the same picture 99.999% of the time when turned on. Even if somebody turns off backlight during the night or dark periods, if turned on during the day it is still true and the same pixels are tensioned the same way always. This will lead to screen picture burn-in after a few years of operation.

**Describe the solution you'd like**
The solution is to "train" the pixels with other contents from time to time. Display of other pages is not enough because the contents of them change in very little, background/button colors are mostly the same, not to speak about a header or a footer page navigation bar which is the same across all pages.

The suggested solution would consist of a function which displays random R,G,B colored pixels all across the screen every 1 second. This could be activated by an MQTT command `antiburn 1` and stopped by touching the screen or by `antiburn 0`. The command could be issued by the HA system in idle periods like when presence reporting nobody's home or during the night or whatever. It would be up to the users when they'd want to trigger this, openHASP should never run this on its own.

**Describe alternatives you've considered**
Use static BMP loaded to flash and looping through one by one. Not good because pictures are always the same (not truly random), flash space eaten up by uncompressed BMP, lot of overhead.

**Additional context**
Not applicable.

RudyFiero, 2021-04-09:
If the backlight is turned off, this would be a good opportunity to vary the image. 

At work I have seen this problem on returned units we manufactured a couple a couple of years ago. We started using color LCDs in our products a few years ago. Another issue is back light life. Many displays are rated for 20,000 hours. On cheap displays they don't even have a rating.

fvanroie, 2021-04-09:
I haven't seen this *feature* in any product, commercial or hobby.

One of the key *features* of openHASP is that the user can display any layout he chooses.
So, you already can display a fullscreen object and alternate the colors to "train" the pixels.

nagyrobi, 2021-04-09:
This is usually included in professional signage displays.

RudyFiero, 2021-04-09:
https://en.wikipedia.org/wiki/Image_persistence
Liquid crystals have a natural relaxed state. When a voltage is applied they rearrange themselves to block certain light waves. If left with the same voltage for an extended period of time (e.g. displaying a pointer or the Taskbar in one place, or showing a static picture for extended periods of time), the liquid crystals can develop a tendency to stay in one position. This ever so slight tendency to stay arranged in one position can throw the requested color off by a slight degree, which causes the image to look like the traditional "burn-in" on phosphor based displays. In fact, the root cause of LCD image persistence is the same as phosphor burn-in, namely, non-uniform usage of the display's pixels.

https://www.adkom.de/en/files/displaytechnik/tft-displays/image-sticking-at-lcds_tfts.pdf

https://www.philips.ca/c-f/XC000007486/what-is-image-sticking,-image-burn-in,-an-after-image,-or-a-ghost-image-on-an-lcd
What is Image Sticking, Image Burn-in, an After Image, or a Ghost Image on an LCD?
Uninterrupted display of still or static images over an extended period may cause “burn in”, also known as “after-imaging” or “ghost imaging”, on your screen. “Burn-in”, “after-imaging”, or “ghost imaging” is a well-known phenomenon in LCD panel technology. In most cases, the “burned in”, “after-imaging”, or “ghost imaging” will disappear gradually over a period of time after the power has been switched off.
Always activate a moving screen saver program when you leave your monitor unattended.

https://www.buydisplay.com/download/manual/ERC12864-14_Datasheet.pdf
8.1 What is Image Sticking?
If you remain a fixed image on OLED Display for a long period of time, you may experience a phenomenon called Image
Sticking. Image Sticking - sometimes also called “image retention” or “ghosting”- is a phenomenon where a faint outline
of a previously displayed image remains visible on the screen when the image is changed. It can occur at variable levels of
intensity depending on the specific image makeup, as well as the amount of time the core image elements are allowed to
remain unchanged on the screen. In POS applications, for example, a button menu which remains fixed, or in which the
“frame” elements (core image) remain fixed and the buttons may change, may be susceptible to image sticking. It is
important to note that if the screen is used exclusively for this application, the user may never notice this phenomenon since
the screen never displays other content. `It is only when an image other than the “retained” image is shown on the screen
that this issue becomes evident. Image sticking is different that the “burn-in” effect commonly associated with phosphor
based devices.

fvanroie, 2021-04-09:
Yes... I did indeed google the term. So, this feature request actually applies to any product using a TFT.

You can already display a screensaver if you want to. Or cycle through R, G, B, white and black yourself.
Make a moving clock...

nagyrobi, 2021-04-09:
You mean by sending MQTT commands for each cycle? For moving pixels every second I need an MQTT call every second...

fvanroie, 2021-04-09:
Sure, if you want to you can update very second.

RudyFiero, 2021-04-09:
That would be acceptable.

nagyrobi, 2021-04-10:
This is how some professional screens to burn-in prevention:
https://displays.agneovo.com/gl/innovations/anti-burn-in%E2%84%A2
See Frame Movement, Line Movement, Dot Movement at the bottom of the article.

nagyrobi, 2021-04-10:
Another way to reduce burn-in would be to clear the screen (display white) when backlight is off. And re-draw when backlight comes back on.
One can of course make an automation for this in the HA system, but I think this could be handled so much faster directly by the MCU.

nagyrobi, 2021-04-10:
Added [example automations](https://haswitchplate.github.io/openHASP-docs/#custom-component/sampl_autom/#prevent-burn-in-of-the-lcd-screen) to achieve some results using HA.

fvanroie, 2021-04-10:
Isn't it easier to just show/hide the overlay object instead of creating a new one and delete?

nagyrobi, 2021-04-10:
why keep it in the memory?

fvanroie, 2021-04-10:
Because there is a 100% chance you'll be needing it again... creating/deleting objects takes more work for the GUI.
But anyways... whatever works!

nagyrobi, 2021-04-22:
Although would be nice to have this as a built-in feature, results are good with automations from HA system also.

fvanroie, 2021-04-22:
Cool, as it would be a long while before this feature can be considered a priority.
Let's revisit when the firmware is stable and more feature-complete.

fvanroie, 2021-09-27:
Please check out and test 1d7e93be830f690babd726aa164ffce05fb63204 in the master branch.

fvanroie, 2021-10-10:
I have decoupled the `antiburn` function from the `backlight`. It's not exposed as an official command yet, but now it is super easy to add this code in the `custom_topic_payload` function:

```cpp
   if(!strcmp(topic, "antiburn")) {
        bool state = Parser::is_true(payload);    // ON, TRUE, YES or 1
        hasp_set_anti_burn(state ? 25 : 0, 1000); // ON = 25 color cycles of 1000 milli seconds (i.e. 25 sec)
    }
```
Then send `on` or `off` to `hasp/<plate>/custom/antiburn`.

When the plate is touched:
- if antiburn was running it will stop, the touch is ignored
- if the backlight was off it will turn on, the touch is ignored

It needs some proper testing, but this seems to be a good solution for those who really want/need this feature.
If all is OK we can decide to make a proper command later... What do you think?

fvanroie, 2021-10-10:
If it goes into the CC, it's best to make it a proper command instead of a custom compile-time option.

dgomes, 2021-10-10:
on a 2nd thought, this is a one time configuration option (does not make sense to expose a switch)

fvanroie, 2021-10-12:
Maybe we can set [`disabled_by_default`](https://developers.home-assistant.io/docs/entity_registry_disabled_by/) on the entity, so it won't be present in the UI by default. But a user who wants to be able to enable it manually can show the switch. When toggled it will enable/disable the cycle.

And when the cycle ends it will automatically turn back off...

nagyrobi, 2021-10-14:
If turned on, will it stop by itself after a while? If not, we need some kinda toggle button for the webif.

fvanroie, 2021-10-18:
@dgomes Do you think it is possible to expose this topic as a switch in HA?
Preferably set disabled_by_default on the entity, since I think most users don't *really* need it in Lovelace...

dgomes, 2021-11-06:
added to custom component in https://github.com/HASwitchPlate/openHASP-custom-component/commit/24c5a7db81251deab97099dd6333d217637adced

fvanroie, 2021-11-06:
To-do: needs to also accept json syntax `{"state": "on"}`

closed 2021-11-17 by nagyrobi