HASwitchPlate/openHASP #136: Popup object for events

MatijaTerzic, 2021-04-12:
**Is your feature request related to a problem? Please describe.**
No

**Describe the solution you'd like**
For example when in Home Assistant a state changes from on / off (eg. lights) a popup is show with custom text and after x ms goes away.

**Describe alternatives you've considered**
NA

**Additional context**
NA

nagyrobi, 2021-04-20:
Also with [page attributes](https://haswitchplate.github.io/openHASP-docs/#pages/#page-attributes) should be possible to create "hidden" pages which don't ever appear during normal page navigation, but can be shown by MQTT commands - thus achieving a behavior of pop-up screens.

fvanroie, 2021-04-21:
@MatijaTerzic Does this sufficiently answer your question?

MatijaTerzic, 2021-04-21:
@fvanroie yes thank you

MatijaTerzic, 2021-04-21:
Ok

closed 2021-04-21 by MatijaTerzic