HASwitchPlate/openHASP #137: Lanbon L8 Display Settings "Long Idle Setting Not Retained"

juan11perez, 2021-04-13:
- Lanbon L8 Switch
- Firmware: lanbon_l8_full_8MB_v0.4.0.bin
- Home Assistant custom component

### Describe the bug
Long idle setting in "Display Settings" is not retained, resulting in the display remaining on indefinitely.

### To Reproduce
This happens after home assistant restarts. Not always immediately, but eventually after 3 or 4 restarts the setting is lost.

### Expected behaviour
Setting retained.

nagyrobi, 2021-04-13:
Try a newer build from https://github.com/HASwitchPlate/openHASP/actions

juan11perez, 2021-04-13:
@nagyrobi  thank you 
short idle 60s
long idle 120s

ok , ive installed the latest and will advise.
I noticed this version already has the relay state at hasp/platexx/state/relay12 
thank you.
But whats the command topic?
tried hasp/plate01/command/relay12 with payload 1 or 0 but no luck.

fvanroie, 2021-04-13:
Are you using the [openHASP custom component](https://github.com/HASwitchPlate/openHASP-custom-component) or some automation to control the dimming of the backlight on idle?

The openHASP firmware doesn't dim the screen by itself, so there is an outside entity sending the `dim` commands.
My guess is that when you reboot HA while the plate is in state `idle long`, the states get out-of-sync.

Try connecting to the plate using telnet and reboot HA a couple of times to see what shows up in the logs.

juan11perez, 2021-04-14:
@fvanroie Thank you 
I am using the custom component and the sample automation which dims the backlight after a few minutes.
I'll try your suggestion and revert.

As mentioned I noticed you now added a state topic for the relay but the previous output command tonturnnit on off doesn't work anymore.
Can you tell me the new topic to turn the relay on or off?
Thank you

fvanroie, 2021-04-14:
I'll check with @dgomes is we need to implement an `idle` command to get the currect idle state of the plate.
That way the custom component can retrieve it when HA reboots. The current idle state can be requested via `statusupdate` but this might not be as straight-forward.

The issue regarding the gpio outputs will be followed up in #40. I'm trying to keep this issue list topical.
Keep in mind that v0.4.1 from the actions tab is a work-in-progress version and thing can (will) be broken in these *nightly* builds.

juan11perez, 2021-04-14:
@fvanroie looking at the automation, i think it's my mistake. below it's the automation. 
effectively when elevation is up i'm sending idle2 to zero.
apologise for wasting your time.

`- id: lights_lanbon_night_day
  alias: Lights lanbon night day
  # initial_state: true  
  trigger:
  - platform: numeric_state
    entity_id: sun.sun
    attribute: elevation
    below: -1
  - platform: numeric_state
    entity_id: sun.sun
    attribute: elevation
    above: 1    
  condition:
  - condition: template
    value_template: "{{ (as_timestamp(now()) - as_timestamp(states('sensor.hass_uptime'))) / 60 > 2 }}"
  action:
  - choose:
    - conditions: '{{state_attr("sun.sun","elevation") > 1}}'
      sequence:   
      - service: mqtt.publish
        data:
          topic: hasp/plate01/config/gui
          payload: '{"idle2":0}'
    default:
    - service: mqtt.publish
      data:
        topic: hasp/plate01/config/gui
        payload: '{"idle2":120}'`

understand on the nightly build. I'm actually not using the switch in it's intended place as I need the relay control.
It's on a bench for testing, so i'll wait until relay is controllable.

fvanroie, 2021-04-15:
Has this issue been resolved and can this issue be closed?

juan11perez, 2021-04-15:
@fvanroie 
Sorry yes. Thank you

closed 2021-04-15 by juan11perez

fvanroie, 2021-04-15:
No worries, glad this has been fixed! 😃

vampywiz17, 2024-05-17:
Hello there, sorry to open this issue, but i have same problem. When i restart HA the backlight can not disable, only if i restart my Lanbon L8. Any idea why to do this?

juan11perez, 2024-05-18:
@vampywiz17 
Did you recently upgrade the openhasp hacs component  to 7.2?
If yes it may be the cause.
I upgraded  to 7.2 and my backlights started acting faulty.
I went back to 6.6, which is where my plates are and issue solved.

vampywiz17, 2024-05-18:
@juan11perez 

Yes! it happen, when i update it to 7.2!

try to downgrade it.