HASwitchPlate/openHASP #138: [MQTT] Main topic length limitation and definition

jsponz, 2021-04-13:
Hi,

Is there a reason why the length of "HASP Node Name" is limited to 16 characters? I would need it longer. 

What about making configurable the main topic "hasp/" ? That would add more flexibility to this nice software!

Thanks,
Jorge

fvanroie, 2021-04-13:
Both of these can be adjusted, but you will need to change compile the firmware yourself.

"hasp/" is configurable via  `MQTT_PREFIX` in the user_config_override.h file.

The node name length has no such configuration option, so you need to update it in the appropriate places.
You say 16 is not enough, but I'm missing why you need longer and what would be an acceptable default length?

jsponz, 2021-04-13:
Thanks for your quick reply. 

Regarding main topic prefix, it would be great to have the option in the web panel, just a suggestion. 

Regarding length, I use the following format to identify my devices: MAC address (12 characters) + scheme ID (6 characters).  

Based on your comment, I can compile my own version supporting 18 characters without any issue. 

Thanks again!

fvanroie, 2021-04-16:
Cool, thanks.

Do you still have an issue or does this answer your question?

jsponz, 2021-04-18:
It is fine. Thanks a lot for your help and your development!

closed 2021-04-18 by jsponz