HASwitchPlate/openHASP #140: \n in a JSON string is not valid

nagyrobi, 2021-04-16:
### Perform all steps below and tick them with [x]
- [x] Check the related part of the [Documentation](https://haswitchplate.github.io/openHASP-docs/)
- [x] Update openHASP to the latest version
- [x] Reproduce the issue and describe all steps

### Describe the bug

openHASP returns the state of dropdowns and rollers containing unescaped newline (`\n`) character.
This is [invalid](https://www.freeformatter.com/json-escape.html) and causes an exception in Home Assistant.

### To Reproduce

Using dropdowns to list the sources and sound modes of a media player.
This is the related starter config loaded config.jsonl from flash:

```json
{"page":6,"id":5,"obj":"dropdown","x":5,"y":128,"w":120,"h":30,"options":"Source1\nSource2\nSource3"}
{"page":6,"id":6,"obj":"dropdown","x":130,"y":128,"w":75,"h":30,"options":"Jazz\nPop\nRock"}
```
The objects get updated correctly by the custom component using templates generating lists separated by newline character:

```yaml
      - obj: "p6b5" # sources list
        properties:
          "options": >
            {% if not (is_state('media_player.hang_up2stream','unavailable')) %}{{"(nincs forrás)\n"|e}}{%for source in state_attr('media_player.hang_up2stream','source_list')%}{{source+"\n"|e}}{%-if not loop.last%}{%-endif%}{%-endfor%}{% endif %}

      - obj: "p6b6" # sound modes list
        properties:
          "options": >
            {% if not (is_state('media_player.hang_up2stream','unavailable')) %}{%for soundmode in state_attr('media_player.hang_up2stream','sound_mode_list')%}{{soundmode+"\n"|e}}{%-if not loop.last%}{%-endif%}{%-endfor%}{% endif %}

```
The objects get populated correctly with the options, however when the plate returns them in MQTT state topics `hasp/plate/state/p6b5` and `hasp/plate/state/p6b6` Home Assistant registers error message _json.decoder.JSONDecodeError: Invalid control character_ at the location of the first newline character:

```text
2021-04-16 17:37:23 ERROR (MainThread) [homeassistant.util.logging] Exception in message_received when handling msg on 'hasp/plate/state/p6b5': '{"options":"(nincs forrás)
3,5 Jack
Bluetooth
USB
PC Audio
Jazzy
Jazzy Robi
Rádió 1
Retró
Sláger
Best FM
Forrás
Manna
Info Rádió
Jazzy Soul
Jazzy Grrove
Jazzy Cool
Petőfi
SpotifyPlayer
Karácsony
Kossuth
Impuls
SubasioPiu"}'
Traceback (most recent call last):
  File "/config/custom_components/openhasp/__init__.py", line 559, in message_received
    message = HASP_EVENT_SCHEMA(json.loads(msg.payload))
  File "/usr/local/lib/python3.8/json/__init__.py", line 357, in loads
    return _default_decoder.decode(s)
  File "/usr/local/lib/python3.8/json/decoder.py", line 337, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
  File "/usr/local/lib/python3.8/json/decoder.py", line 353, in raw_decode
    obj, end = self.scan_once(s, idx)
json.decoder.JSONDecodeError: Invalid control character at: line 1 column 27 (char 26)

2021-04-16 17:37:23 ERROR (MainThread) [homeassistant.util.logging] Exception in message_received when handling msg on 'hasp/plate/state/p6b6': '{"options":"Classic
Jazz
Normal
Pop
Vocal"}'
Traceback (most recent call last):
  File "/config/custom_components/openhasp/__init__.py", line 559, in message_received
    message = HASP_EVENT_SCHEMA(json.loads(msg.payload))
  File "/usr/local/lib/python3.8/json/__init__.py", line 357, in loads
    return _default_decoder.decode(s)
  File "/usr/local/lib/python3.8/json/decoder.py", line 337, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
  File "/usr/local/lib/python3.8/json/decoder.py", line 353, in raw_decode
    obj, end = self.scan_once(s, idx)
json.decoder.JSONDecodeError: Invalid control character at: line 1 column 20 (char 19)

2021-04-16 17:37:25 ERROR (MainThread) [homeassistant.util.logging] Exception in message_received when handling msg on 'hasp/plate/state/p6b5': '{"options":"(nincs forrás)
3,5 Jack
Bluetooth
USB
PC Audio
Jazzy
Jazzy Robi
Rádió 1
Retró
Sláger
Best FM
Forrás
Manna
Info Rádió
Jazzy Soul
Jazzy Grrove
Jazzy Cool
Petőfi
SpotifyPlayer
Karácsony
Kossuth
Impuls
SubasioPiu"}'
Traceback (most recent call last):
  File "/config/custom_components/openhasp/__init__.py", line 559, in message_received
    message = HASP_EVENT_SCHEMA(json.loads(msg.payload))
  File "/usr/local/lib/python3.8/json/__init__.py", line 357, in loads
    return _default_decoder.decode(s)
  File "/usr/local/lib/python3.8/json/decoder.py", line 337, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
  File "/usr/local/lib/python3.8/json/decoder.py", line 353, in raw_decode
    obj, end = self.scan_once(s, idx)
json.decoder.JSONDecodeError: Invalid control character at: line 1 column 27 (char 26)

2021-04-16 17:37:25 ERROR (MainThread) [homeassistant.util.logging] Exception in message_received when handling msg on 'hasp/plate/state/p6b6': '{"options":"Classic
Jazz
Normal
Pop
Vocal"}'
Traceback (most recent call last):
  File "/config/custom_components/openhasp/__init__.py", line 559, in message_received
    message = HASP_EVENT_SCHEMA(json.loads(msg.payload))
  File "/usr/local/lib/python3.8/json/__init__.py", line 357, in loads
    return _default_decoder.decode(s)
  File "/usr/local/lib/python3.8/json/decoder.py", line 337, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
  File "/usr/local/lib/python3.8/json/decoder.py", line 353, in raw_decode
    obj, end = self.scan_once(s, idx)
json.decoder.JSONDecodeError: Invalid control character at: line 1 column 20 (char 19)

```
### Expected behavior

Jsons generated by the plate in MQTT payloads need to be checked not to contain invalid characters.

### Screenshots or video
The plate works fine otherwise:
[image]

dgomes, 2021-04-16:
\n should be escaped, or better yet, replaced with ","

fvanroie, 2021-04-16:
I was aware that quotes in the string might mead to issues, apparently there's more characters to need special treatment.
Escaping strings is always though, replacing characters is also problematic because you can have a list that already contains comma's...

The best way is the run all returned JSON objects containing strings through the ArduinoJson encoder.
There will be some overhead, but at least the result is guaranteed to be json compliant without needing much code.

fvanroie, 2021-04-16:
Pushed to master

nagyrobi, 2021-04-17:
Fix works thanks.

closed 2021-04-17 by nagyrobi