HASwitchPlate/openHASP #143: wt32 sc01 v5.0 cant update

xHirscHx, 2021-04-19:
So tried to update to v5.0 and i get this when i use  1. esphome and this when i use  2.  tasmo.
1
Using 'COM6' as serial port.
Connecting....
Detecting chip type... ESP32
Connecting....

Chip Info:
 - Chip Family: ESP32
 - Chip Model: ESP32D0WDQ5 (revision 1)
 - Number of Cores: 2
 - Max CPU Frequency: 240MHz
 - Has Bluetooth: YES
 - Has Embedded Flash: NO
 - Has Factory-Calibrated ADC: YES
 - MAC Address: 24:6F:28:D2:61:C8
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
 - Flash Size: 4MB
Unexpected error: The firmware binary is invalid (magic byte=FF, should be E9)
2
Invalid head of packet (0x46)

fvanroie, 2021-04-19:
The full binary is meant to be flashed with [ESPtool](https://haswitchplate.github.io/openHASP-docs/#ESP32/), [Flash Download Tools (ESP8266 & ESP32 & ESP32-S2)](https://www.espressif.com/en/support/download/other-tools) or [Tasmota-PyFlasher](https://github.com/tasmota/tasmota-pyflasher/releases/tag/1.0) :

You need to write the full binary to the ESP32 flash chip, starting at address `0x0`:
   [image]
Make sure to select the correct com port.

Tasmota-PyFlasher also works:
[image]

The ota binary can be uploaded via the web UI, if the plate has been previously flashed with openHASP 0.4.0.

The ESPhome-Flasher expects the ota binary to be written to address 0x10000. You can flash the ota firmware directly using ESPhome-Flasher, but **the ESP only boots correctly if the bootloader and partitiontable are already present.**

xHirscHx, 2021-04-19:
Oh ya sorry brain fart. Mix them up  again sorry.

closed 2021-04-20 by xHirscHx