HASwitchPlate/openHASP #145: Feature request: GPIO info on request

nagyrobi, 2021-04-20:
**Is your feature request related to a problem? Please describe.**
To verify the status of the GPIO settings from HA side. At boot HA could check the plate's gpio settings and if finds them inappropriate, push down the right config.

**Describe the solution you'd like**
When publishing an empty payload to `hasp/plate/config/gpio` it would be nice if the plate would respond with current GPIO settings in effect. Optionally put it in `state/statusupdate`.

**Describe alternatives you've considered**
n/a

**Additional context**
n/a

nagyrobi, 2021-04-20:
Edit: on Lanbon L8 pushing `{"config":[2360346,2491680,2623009,2097420,2097678,2097947,0,0]}` to `hasp/<nodename>/config/gpio` doesn't work on 0.5.0.

fvanroie, 2021-04-21:
> Edit: on Lanbon L8 pushing {"config":[2360346,2491680,2623009,2097420,2097678,2097947,0,0]} to hasp/<nodename>/config/gpio doesn't work on 0.5.0.

See #146

closed 2021-04-22 by nagyrobi