HASwitchPlate/openHASP #146: Submodule config get/set not working

nagyrobi, 2021-04-20:
### Perform all steps below and tick them with [x]
- [x] Check the related part of the [Documentation](https://haswitchplate.github.io/openHASP-docs/)
- [x] Update openHASP to the latest version
- [x] Reproduce the issue and describe all steps

### Describe the bug
Originally I submitted a [feature request](https://github.com/HASwitchPlate/openHASP/issues/145) to get / set the gpio configuration via MQTT but it turned out that none of the [config/submodule](https://haswitchplate.github.io/openHASP-docs/#commands/#configsubmodule) commands work.

### To Reproduce

Publish and empty paylod to `hasp/<nodename>/config/gui`. No response.
Same experience for all [config/submodules](https://haswitchplate.github.io/openHASP-docs/#commands/#configsubmodule).

Go to web interface > Configuration > Display Settings. Check the value for `Long Idle`, make sure it's not 224.
Publish payload `{"idle2":224}` to `hasp/<nodename>/config/gui`. Watch the debug console, nothing happens.

### Expected behavior
Check the Check the value for `Long Idle` in the web interface again by refreshing the page. Should have been changed, but it didn't.
Empty payloads to `hasp/<nodename>/config/gui` should result in responses published to `hasp/<nodename>/state/config`

### Screenshots or video
n/a

MatijaTerzic, 2021-04-21:
Also output[x] does not work anymore:

 MSGR: Command 'output[1]' not found => 1

fvanroie, 2021-04-21:
`config/submodule` act as normal commands and are also accepted on `hasp/<nodename>/command/config/submodule`.
Dunno if it makes sense to keep a separate `config` topic for those.

nagyrobi, 2021-04-21:
Just tried both:
```
hasp/<nodename>/command/config/gui 
hasp/<nodename>/config/gui 
```
No response on either.

closed 2021-04-22 by nagyrobi