HASwitchPlate/openHASP #152: Iobroker get no events from slider,switch,dropdown,togglebutton

freshnas, 2021-04-23:
I use a Lanbon L8 with the Firmware lanbon_l8_full_8MB_v0.5.0.bin.
My Demo pages.jsonl is:

{"page":1,"comment":"---------- Page 1 ----------"}
{"obj":"btn","id":1,"x":10,"y":45,"w":220,"h":55,"toggle":true,"text":"Push Me \uf0a6"}
{"obj":"cb","id":2,"x":10,"y":100,"w":220,"h":55,"text":"A Checkbox"}
{"obj":"label","id":3,"x":10,"y":10,"w":220,"h":30,"text":"My Label","align":1,"padh":50}
{"obj":"switch","id":4,"x":70,"y":205,"w":100,"h":55}
{"obj":"led","id":5,"x":10,"y":205,"w":55,"h":55}
{"obj":"dropdown","id":6,"x":10,"y":150,"w":150,"options":"\uf007 Line 1\n\uf007 Line 2\n\uf007 Line 3"}
{"obj":"spinner","id":7,"x":165,"y":140,"w":70,"h":70}

{"page":2,"comment":"---------- Page 2 ----------"}
{"obj":"slider","id":1,"x":10,"y":170,"w":200,"h":50}
{"obj":"gauge","id":3,"x":13,"y":10,"w":100,"h":100}
{"obj":"bar","id":4,"x":10,"y":110,"w":200,"h":50}
{"obj":"lmeter","id":5,"x":127,"y":10,"w":100,"h":100}
{"obj":"label","id":10,"x":0,"y":0,"w":70,"h":50,"parentid":5,"text":"\uf00c OK"}

{"comment":"---------- Page 3 ----------"}
{"obj":"cpicker","id":1,"page":3,"x":20,"y":0,"w":200,"h":200}
{"obj":"cpicker","id":2,"page":3,"x":20,"y":210,"w":200,"h":50,"rect":true}

{"page":4,"comment":"---------- Page 4 ----------"}
{"obj":"dropdown","id":1,"x":10,"y":10,"w":220,"options":"Spring\nSummer\nAutumn\nWinter"}
{"obj":"roller","id":2,"x":40,"y":50,"w":160,"rows":3,"options":"2020\n2021\n2022\n2023\n2024"}

{"page":0,"id":1,"obj":"btn","action":"prev","x":0,"y":290,"w":79,"h":32,"bg_color":"#2C3E50","text":"\uE141","text_color":"#FFFFFF","radius":0,"border_side":0,"text_font":28}
{"page":0,"id":2,"obj":"btn","action":"back","x":80,"y":290,"w":80,"h":32,"bg_color":"#2C3E50","text":"\uE2DC","text_color":"#FFFFFF","radius":0,"border_side":0,"text_font":22}
{"page":0,"id":3,"obj":"btn","action":"next","x":161,"y":290,"w":79,"h":32,"bg_color":"#2C3E50","text":"\uE142","text_color":"#FFFFFF","radius":0,"border_side":0,"text_font":28}

Iobroker dont show events or values from slider,switch,dropdown,togglebutton

[image]

When i use the MQTT-Explorer, i can see a value receive from the Objects, but without events .

[image]

But when i use Openhab, i receive a correct event with value

[image]
[image]

The Mqtt-Explorer receive a correct event and value too with openhab
[image]

What the problem with Iobroker???
Thanks!!!

fvanroie, 2021-04-23:
This seems to be an issue of the iobroker.mqtt server: This [piece of code](https://github.com/ioBroker/ioBroker.mqtt/blob/7d90662f1a7212f928ccea54672f4988a18da955/lib/server.js#L537-L558) handles the incoming messages.

They parse the payload using a *Fast solution*, resulting in false positives. It is even mentioned in that code what the *right solution* is...

If the payload is a json object containing the `val` property, it uses the `val` property only as value of that object...

freshnas, 2021-04-24:
The change in server.js solved the problem.
Thank you for the fast support!!!!

closed 2021-04-24 by freshnas

fvanroie, 2021-04-24:
You are welcome. I'm glad that workaround fixes it.