HASwitchPlate/openHASP #168: Add MDI icons for weather conditions

nordeep, 2021-05-25:
**Is your feature request related to a problem? Please describe.**
It is convenient to display the current and feature weather conditions by MDI icons. But at this moment not all conditions are covered by compiled MDI font.

**Describe the solution you'd like**
Add next list of glyphs from materialdesignicons-webfont.ttf:
weather-cloudy - 0F0590
weather-fog - 0F0591
weather-hail - 0F0592
weather-lightning - 0F0593
weather-night - 0F0594
weather-partly-cloudy - 0F0595
weather-pouring - 0F0596
weather-rainy - 0F0597
weather-snowy - 0F0598
weather-sunny - 0F0599
weather-windy - 0F059D

**Describe alternatives you've considered**

**Additional context**

nagyrobi, 2021-05-25:
We plan to use a separate uploadable .zi font for these, as including all of them in all font sizes would consume too much firmware flash space.

fvanroie, 2021-05-26:
The weather icons were discussed but ultimately they were not included due to limited flash size.
To properly support weather icons, there are over a dozen icons needed...
We opted to include ~120 more generic home automation icons instead.

There are still ways to get weather icons though:
- Build and include your own custom font
- Use the new `img` object in 0.6
- Use an uploadable .zi font

fvanroie, 2021-06-27:
For 0.6.1 I'm testing the inclusion of these weather icons. The flash space is getting quite cramped on the 4MB chips with large displays:

```
d1-mini-esp8266_ili9341: [========  ]  76.2% (used 796392 bytes from 1044464 bytes)
d1-mini-esp32_ili9341:   [========= ]  93.6% (used 1595410 bytes from 1703936 bytes)
wt32-sc01:               [========= ]  92.8% (used 1581246 bytes from 1703936 bytes)
huzzah32-featherwing-35: [==========]  96.8% (used 1649594 bytes from 1703936 bytes)
huzzah32-featherwing-24: [========= ]  94.0% (used 1602134 bytes from 1703936 bytes)
freetouchdeck:           [==========]  96.2% (used 1638510 bytes from 1703936 bytes)
esp32-touchdown:         [==========]  96.6% (used 1645850 bytes from 1703936 bytes)
yeacreate-nscreen32:     [==========]  96.2% (used 1639358 bytes from 1703936 bytes)
m5stack-core2:           [========  ]  81.8% (used 1715366 bytes from 2097152 bytes)
lanbon_l8:               [========  ]  82.4% (used 1728818 bytes from 2097152 bytes)
```

nagyrobi, 2021-06-27:
Is it worth it? Using discrete colorful png images of any size from the fs give more [flexibility](https://haswitchplate.github.io/openHASP-docs/0.6/integrations/home-assistant/sampl_conf/#current-weather-and-forecasts).

(A cloud with a magnifying glass (search) would still be needed though, for air quality sensors)

fvanroie, 2021-06-27:
Each user has another set of icons which are deemed *needed*... there is no right or wrong here.
At some point you need to choose what's more important to put in the default firmware or compile your own.

nagyrobi, 2021-06-27:
That was the reason we left out weather icons in the first place. Weather is a very sensitive topic because is highly location-specific, not to speak about user's aesthetical demands. The icons in the MDI set are only a few of the possible weather conditions around the world. 
There are tons of free weather icons out there in various image formats, the example above shows how to make a weather widget with HA's most basic weather integrations out of the box. Extending the example with any other extra weather conditions and corresponding icons it's very straightforward and does not need any firmware recompilation etc.

nagyrobi, 2021-07-04:
mdi-cloud-search-outline

fvanroie, 2021-07-04:
```
d1-mini-esp8266_ili9341: [========  ]  76.5% (used 798824 bytes from 1044464 bytes)
d1-mini-esp32_ili9341:   [========= ]  93.5% (used 1592414 bytes from 1703936 bytes)
yeacreate-nscreen32:     [========  ]  77.9% (used 1632914 bytes from 2097152 bytes)
lanbon_l8:               [========  ]  82.3% (used 1726066 bytes from 2097152 bytes)
huzzah32-featherwing-24: [========= ]  93.8% (used 1599142 bytes from 1703936 bytes)
esp32-touchdown:         [==========]  96.2% (used 1639422 bytes from 1703936 bytes)
m5stack-core2:           [========  ]  81.7% (used 1713018 bytes from 2097152 bytes)
freetouchdeck:           [==========]  95.8% (used 1632070 bytes from 1703936 bytes)
wt32-sc01:               [========= ]  92.4% (used 1575246 bytes from 1703936 bytes)
huzzah32-featherwing-35: [==========]  96.4% (used 1643134 bytes from 1703936 bytes)
```

`weather-sunny` and `weather-night` were already in v0.6.0

Icons added to v0.6.1-dev:
- cloud-search-outline
- recycle-variant
- weather-cloudy
- weather-fog
- weather-hail
- weather-lightning
- weather-partly-cloudy
- weather-pouring
- weather-rainy
- weather-snowy
- weather-windy
- weather-windy-variant
- weather-lightning-rainy
- weather-snowy-rainy

nagyrobi, 2021-07-05:
Not working:
- F39D  recycle-variant
- E957 cloud-search-outline

fvanroie, 2021-07-11:
These have been added to v0.6.1-dev available from the Actions tab.

closed 2021-07-11 by fvanroie