HASwitchPlate/openHASP #169: Feature request: WebUI and override options for Show IP address on connect

nagyrobi, 2021-05-27:
**Is your feature request related to a problem? Please describe.**
Commit https://github.com/HASwitchPlate/openHASP/commit/57569e7b06676dc175d2875ff8578e4e0075dfde is excellent, thank you!
To make deployments easier, please consider these:
- add an option in web ui configuration to disable `Show IP address on connect` (could be a checkbox which is enabled by default)
- add an option to `user_config_override` to disable `auto_close` of the messagebox.

**Describe the solution you'd like**
The reason for the first: the IP address may be unimportant for the end-user after deployment is finished (wife acceptance factor).
The reason for the second: when multiple devices are deployed, with wifi credentials burned-in with `user_config_override` it takes time to walk by in each room and note the IP addresses gathered, by that time the messageboxes closed. For such cases keep that info on the screen until dismissed, and when configuration is done, first option can be used to disabl popup of the address on the next boot.

**Describe alternatives you've considered**
N/A

**Additional context**
N/A

fvanroie, 2021-08-02:
In v0.6.1 the popup screen with the IP address has been removed.
The `/online.cmd` script will run after the wifi connects.
The idea is that you can implement a message box to show the IP.

nagyrobi, 2021-08-03:
In the next version we need to pre-create the /online.cmd with the msgbox for fresh flashes and factory resets.

fvanroie, 2021-08-03:
#https://github.com/HASwitchPlate/openHASP/commit/d3da4957bb37e408bca1eb842e697614bbd8a522

closed 2021-08-04 by fvanroie