HASwitchPlate/openHASP #174: Safety mode

zsamiatt, 2021-06-10:
If wifi or HA dead need safety mode.
We need control inbuild relay. It's hard to fix problem in the dark. And I think the screen protection should also work without network.

nagyrobi, 2021-06-10:
Ongoing discussion about this: https://github.com/HASwitchPlate/openHASP/discussions/61

fvanroie, 2021-08-02:
> In 0.6.1-dev the automatic reboot on wifi disconnect has been removed. The plate will keep on trying to reconnect indefinitely...
>
> `/online.cmd` and `/offline.cmd` scripts are executed when the wifi status changes. This makes it possible to disable or hide buttons, go to a special offline page, etc...
> 
> Mqtt has not yet received the same update, but will probably get a similar change.

nagyrobi, 2021-08-03:
https://haswitchplate.github.io/openHASP-docs/0.6.1/commands/#batch-processinng

fvanroie, 2022-01-23:
I have created a `feature-mqtt` branch that doesn't obstruct the normal GUI while MQTT reconnects.
The `Connecting...` progressbar is gone and the plate keeps responding as normal to local events.

This needs thorough testing though... as it involves some multi-threading and semaphores, which I'm not really familiar with.
Preliminary testing seems OK here, but just in case you want to test this branch I would not advised to put it on a production plate.

bensuffolk, 2022-05-23:
Can you add a feature to run a cmd file when MQTT connects and when it disconnects like you do for wifi?

bensuffolk, 2022-07-07:
That sounds like a sensible way forward to me.

fvanroie, 2022-07-07:
Now that I think about it... it could have an influence on the ETHSPI code. `PubSub` uses an Arduino `WifiClient` object as its underlying network protocol. I don't think that `esp_mqtt` supports the Arduino networking library since it's an ESP-IDF module. Probably the ESP-IDF W5500 lwip network interface needs to used...

bensuffolk, 2022-07-09:
The ETHSPI is using the lwip network interface, so it will all just work with the WiFiClient. It's not a separate Ethernet library.

fvanroie, 2022-07-09:
My bad! I thought it was a purely Arduino SPI library, but looking under the hood indeed reveals the ESP-IDF framework being used. This is good news and should indeed be supported by the `esp_mqtt` client.

I have checked in the code but now somehow the Github throws an error while I'm able to compile locally...

fvanroie, 2022-07-23:
I've worked around the error by disabling Freetype for the time being.

fvanroie, 2022-11-21:
I've added `mqtt_on.cmd` and `mqtt_off.cmd` scripts.

closed 2024-08-16 by fvanroie