HASwitchPlate/openHASP #177: Feature request: swipe around tabview at ends

nagyrobi, 2021-06-11:
**Is your feature request related to a problem? Please describe.**
Tabview can be used with `"btn_pos":0`. This hides the tab buttons. In this case switching between the tabs can be done by swiping left or right. When on first tab and swiping backwards (to the right), would be good if it would change to last tab. And when on the last tab and swiping forward (to the left) would be nice if it would change to the first tab.

**Describe the solution you'd like**
As above description.

**Describe alternatives you've considered**
N/A

**Additional context**
N/A

fvanroie, 2021-06-11:
I have searched for this functionality in the lvgl docs and the internet. It seems the [feature was proposed](https://github.com/lvgl/lvgl/issues/1604) at some point here but never materialized.

I posted this question on the [lvgl forum](https://forum.lvgl.io/t/circular-tab-view/5786).

fvanroie, 2021-07-02:
There's no easy way to implement this feature in lvgl. So unless somebody wants to look into this further, this won't be implemented in openHASP either.

nagyrobi, 2021-07-02:
Understood.

closed 2021-07-02 by nagyrobi