HASwitchPlate/openHASP #183: Feature request: Button matrix uncheck all

nagyrobi, 2021-07-02:
**Is your feature request related to a problem? Please describe.**
A button matrix with `"toggle":1` and `"one_check":1` will start with all buttons unchecked (not pressed). In the example below the button matrix implements a radio station preset selector.
[image]
After pressing a button, there's no way to release it without having to press another button `val = 2`:
[image]
The use case is that the button should be in pressed state while that radio preset is playing. If the player is stopped (or it's playing something else than the button presets), no radio is playing, thus all buttons should be in released state.

**Describe the solution you'd like**
When setting val to `-1` or some magic string, all the buttons should go unselected.

**Describe alternatives you've considered**
N/A

**Additional context**
Could be useful for dropdown too. Usually a dropdown starts with the first item in the list, but it would be useful to also be able to set an "empty" item in the dropdown, based on the logic above.

nagyrobi, 2021-07-02:
Related discussion: https://github.com/HASwitchPlate/openHASP/discussions/180

fvanroie, 2021-07-02:
It always comes down to what is supported by [the lvgl object](https://docs.lvgl.io/7.11/widgets/btnmatrix.html).

`one_check` only makes sure that there is _maximum_ 1 button checked... the user can't uncheck the same button once checked.
So either custom object behavior is needed or special treatment of existing properties...

Like add a `no_check` option for *pressing the same button by the user again will uncheck it again*?
Or only allow it to be reset programmatically?

nagyrobi, 2021-07-02:
Programatically is enough for my use case.

fvanroie, 2021-07-02:
You can try commit a5b783a, untested but should work...

nagyrobi, 2021-07-04:
Thank you, works perfectly!

closed 2021-07-04 by nagyrobi