HASwitchPlate/openHASP #207: Delay between HA and openHASP can cause loops when not using groupid

Stupco, 2021-08-30:
### Perform all steps below and tick them with [x]
- [X] Check the related part of the [Documentation](https://haswitchplate.github.io/openHASP-docs/)
- [X] Update openHASP to the latest version
- [X] Reproduce the issue and describe all steps

### Describe the bug
When turning on the light, the command is sent from the plate to HA to turn on the light. If you are using the openHASP custom component to set state values for buttons, they can sometimes lag out and cause small loops.

1. Use plate to turn on light, MQTT message to HA, button state changes.
2. HA sees that the light is OFF, so sets button state to OFF, triggering another MQTT message.
3. HA receives the ON command, turns on light, sets button state to ON.

This causes the light to flicker a few times, until it finds a normalised states. Unfortunately it can finally decide to settle in the OFF state, so it is impossible to turn lights on and keep on using the plate in production with this bug. Usually a reset of the plate (remove from wall and reconnect) can resolve this. Will usually occur a few days later.

### To Reproduce
I think this could potentially be caused by a delay caused in WiFi messages and **MAY** be more pronounced the further your plate is away from the WiFi router.
Provide a small, independent code sample that can be used to reproduce the issue.
Format the code like this:
```json
{"comment":" ----------- Light Button ------------"}
{"page":1,"id":2,"obj":"btn","x":5,"y":40,"w":73,"h":73,"toggle":true,"text":"\uE335","text_font":40,"align":1,"shadow_color":"#000000","shadow_width": 16,"shadow_ofs_y":2, "shadow_ofs_x":2}
{"page":1,"id":3,"obj":"slider","x":94,"y":65,"w":126,"h":20, "min":0,"max":255, "opacity": 255,"shadow_color":"#000000","shadow_width": 16,"shadow_ofs_y":2, "shadow_ofs_x":2}
{"page":1,"id":4,"obj":"btn","x":5,"y":40,"w":230,"h":73,"toggle":false,"text":"\uE335","text_font":40,"align":1,"hidden":true, "click":false,"shadow_color":"#000000","shadow_width": 16,"shadow_ofs_y":2, "shadow_ofs_x":2}
```
```HA config
      - obj: "p1b2" #light toggle button
        properties:
          "val": '{{ 1 if states("light.b1_lights") == "on" else 0 }}'
          "text": '{{ "\uE6E8" if is_state("light.b1_lights", "on") else "\uE335" | e }}'
        event:
          "up":
            - service: light.toggle
              data_template:
                brightness: '{{ states("input_number.b1_lights") }}'
              target:
                entity_id: light.b1_lights
      - obj: "p1b3" #slider
        properties:
          "val": '{{ states("input_number.b1_lights") }}'
          "opacity": '{{ "255" if is_state("light.b1_lights", "on") else "150" | e }}'
        event:
          "up":
            - service: input_number.set_value
              target:
                entity_id: input_number.b1_lights
              data_template:
                value: "{{ val }}"
            - service: light.turn_on
              data_template:
                brightness: "{{ val }}"
              target:
                entity_id: light.b1_lights
      - obj: "p1b4" #HIDER
        properties:
          "hidden": '{{ true if states("light.b1_lights") == "on" else false }}'
          "click": '{{ false if states("light.b1_lights") == "on" else true }}'
        event:
          "up":
            - service: light.toggle
              data_template:
                brightness: '{{ states("input_number.b1_lights") }}'
              target:
                entity_id: light.b1_lights
```

### Expected behavior
Lights to not flicker and to turn on.

Noting that the use of GROUPID will remove this issue (I will start to update my jsonl pages), unfortunately until the last brightness is passed through [(issue:99)](https://github.com/HASwitchPlate/openHASP/issues/99) the lights turn on at full brightness which is not a great solution for production use cases.

### Screenshots or video
I will see if I can post a video.

Stupco, 2021-08-30:
Updated with telnet logs.

fvanroie, 2021-08-30:
I think this needs to be looked at from the HA / CC side... there is a ton of repeated commands sent to the plate.
So that's why it is acting erratic.