HASwitchPlate/openHASP #208: groupid to handle dimmer arguments

Stupco, 2021-08-30:
**Is your feature request related to a problem? Please describe.**
No

**Describe the solution you'd like**
Currently, groupid is for simple commands (which I get). However, it only works well for simple relays. Requesting it to provide ability to handle an additional parameter to handle brightness.

As @fvanroie states:
"it's not how groups work though...
the state is based on an event
which is represented as a percentage
on/off is 0%/100%
there is not inbetween"

**Describe alternatives you've considered**
There is no alternative for local brightness control without having to resort to HA.
However, restoring the previous brightness as mentioned #99 might solve the urgent need for this.

fvanroie, 2021-08-30:
You can have a btnmatrix or a slider to control the [L8-HD brightness](https://haswitchplate.github.io/openHASP-docs/0.6.1/configuration/gpio/#dimmable-lights). Object with a range of values can control the brightness proportional to their current value.

On/off controls like toggle buttons, switches and checkboxes only send 0% or 100% values.

Stupco, 2021-09-14:
Workaround fixed with commit https://github.com/HASwitchPlate/openHASP/commit/75c87665bc29c84f49bca9e7d5f3a0de085e5b2b

However, this creates a new issue: #212

closed 2021-09-14 by Stupco