HASwitchPlate/openHASP #212: Cannot turn on Dimmer if no previous brightness

Stupco, 2021-09-14:
### Perform all steps below and tick them with [x]
- [X] Check the related part of the [Documentation](https://haswitchplate.github.io/openHASP-docs/)
- [X] Update openHASP to the latest version
- [X] Reproduce the issue and describe all steps

### Describe the bug
IF turning on light for first time since restart, HA will not turn on light on as there was no previous brightness known.

Occurs after resetting both HA and Plate.

### To Reproduce
## Debug log
HA Log:
```
2021-08-31 16:01:46 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'mappingproxy object' has no attribute 'brightness' when rendering '{{ states.light.b3_lights.attributes.brightness }}'
```
Telnet Log:
```
[08:03:13.770][96656/99960  3][35304/35520  1] MQTT RCV: hasp/b3/command/output12 = {"state": true, "brightness": 0}
[08:03:13.776][96656/98308  1][35304/35520  1] GPIO: Dimmer => ef 02 00 ed
[08:03:13.781][95104/96376  1][35304/35520  1] MSGR: GROUP 1 value 0 (0-255)
[08:03:13.791][93552/94636  1][35304/35520  1] MQTT PUB: output12 => {"state":"off","brightness":0}

### Expected behavior
Light can turn on with no previous brightness state with simple toggle command, recommend maybe setting brightness value to 255 if no previous state found?

Would still prefer to have previous brightness passed when known as provided in https://github.com/HASwitchPlate/openHASP/commit/75c87665bc29c84f49bca9e7d5f3a0de085e5b2b

fvanroie, 2021-09-20:
@Stupco Can you test this again now?

Stupco, 2021-09-21:
Seems to be working much better! But I don't think it exposes the Brightness using this method.

Found the following in the Home Assistant logs to show that, even though I've turned it off/on and set brightness values too, it is not being exposed when toggling (note: this might be a CC issue):
`2021-09-21 11:12:50 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'mappingproxy object' has no attribute 'brightness' when rendering '{{ states.light.b3_lights.attributes.brightness }}'`

nagyrobi, 2021-09-21:
That's by design in HA. A light when turned off, doesn's expose such attributes.

Stupco, 2021-09-21:
> That's by design in HA. A light when turned off, doesn's expose such attributes.

I know this was already mentioned before, just curious as this error does not show up for my Zigbee lights? I know it's a different protocol, just as example.

Anyway, noting that the HA Error Log does not affect actual functionality as the Brightness value is not necessary when the light is off, we could likely close this issue as it seems to be functioning well?

fvanroie, 2021-09-21:
IMHO Since you can change the brightness when the light is off, it's useful to also report the brightness when the light is off.
openHASP can also be used by other automation systems like OpenHAB or NodeRED that do not have this *quirk*.

It should be possible for HA to treat this attribute as optional instead... but that's another discussion.
BTW, it is a warning and not an error, so everything should be fine. Just ignore that annoying message :)

fvanroie, 2021-09-23:
I'm closing this issue as it is working as expected.
In case of issues, feel free to reopen.

closed 2021-09-23 by fvanroie