HASwitchPlate/openHASP #246: Updates for Responsive web UI

fvanroie, 2021-11-01:
**Is your feature request related to a problem? Please describe.**
The current web UI is clunky and looks old fashioned.

**Describe the solution you'd like**
Implement modern technologies like CSS variables, JavaScript and responsive document queries.
Use a minimalistic approach and include the sources as a gzipped document in the firmware.
These static CSS/JS files can be cached in the web-browser, leading to faster responses for each page.

**Describe alternatives you've considered**
Using standard templating engines would bloat the firmware size with unnecessarily big JavaScript libraries, or we would have to link to them on the internet, which would not work offline or in a restricted network environment.

**Additional context**
- [ ] Add `style.css` and `vars.css`. Gzipped in firmware which can be overridden in flash.
- [ ] Add `script.js`. Gzipped in firmware which can be overridden in flash.
- [ ] Update forms to use simple html code, leaving the styling to the CSS.
- [ ] Use JavaScript to render JSON objects in the browser instead of serving full HTML from the MCU.

nagyrobi, 2021-11-08:
Some results:
On mobile:
[image]
(needs buttons wider+centered)

[image]
(GPIO settings need brushing)

[image]
(same on desktop)

fvanroie, 2021-11-16:
This is mostly complete.

GPIO forms need some work, but I will do these when updating that module with sensors later.
AP-mode has not yet been tested...

kquinsland, 2022-01-27:
This came from a discussion on Discord. 

I am currently working on a tool that - eventually - should be able to connect to a device and get a dump of all of it's state: every detail! What GPIO(s), syslog settings, is the telnet server on... mqtt user/pass/host ... etc.

The ultimate goal is to flash a device with openHasp and configure WiFi via UART before re-assembling / deploying the device in its intended location.

Once the device has an IP address, I should be able to run a simple tool that will:

- set mqtt info
- turn off Telnet
- turn off mdns
- set the syslog endpoint
- adjust brightness / timeouts
- adjust the default start page
- configure GPIOs
- sets the default theme
- ... etc.
- upload misc `jsonl` and `cmd` files
- Save all the changes and reboot

Basically automate all the clicking I'd do normally and ensure that the declared state and the actual device state _match_ always.

I was having a miserable time trying to get this via telnet... at least partly because the telnetlib that python ships with does not appear to easily support "interactive" usecases like mine.  I pivoted to  HTML as scraping it _does_ offer a predictable payload that can be parsed to get device info and device state - including files - can be managed this way.

I am able to get _most_ of the device information with basic regex parsing but a more machine readable format would be simpler.

so instead of having to:

```python3
wifi_cfg_raw = urllib.get(http://1.2.3.4/config/wifi)
wifi_creds = re.match(.wifi_cfg_raw, r'...' )
```

I could just do this

```python3
dev_cfg_doc_str = urllib.get(http://1.2.3.4/api/config.json)

dev_cfg = json.parse(dev_cfg_doc_str)

// set new wifi name
dev_cfg['wifi']['ssid'] = 'someNetwork'

# or POST or whatever
urllib.patch(http://1.2.3.4/api/config.json, dev_cfg)
```

Depending on exactly how the web UI is revamped, it could be possible to have HTML make `PATCH` calls to `/config.json` or similar rather than the current method:

```<form method="POST" action="/config">...</form>```

It may also be easier to implement an API that works a bit more like the telnet interface: rather than `PATCH` calls to a singe file, a series of endpoints is implemented to 'mirror' the telnet:

to get/set GPIO config:

```shell
$ telnet 1.2.3.4
<...>
Prompt > config/gpio
[15:58:34.999][65524/115604 43][42552/42872  1] MSGR: config/gpio
[15:58:35.022][65524/114740 42][42552/42872  1] CONF: {"config":[197658,263456,329249,655628,655886,656155,0,0]}
[15:58:35.048][65524/115208 43][42552/42872  1] MQTT PUB: config => {"gpio":{"config":[197658,263456,329249,655628,655886,656155,0,0]}}

Prompt > config/gpio {"gpio":{"config":[1,1,1,1,1,1,0,0]}}
```

So the `/config/gpio` page would load some bootstrap HTML/JS which would _then_ send a `GET` to `http://1.2.3.4/api/config/gpio` and would get back a simple JSON doc like:

 ```
{"gpio":{"config":[197658,263456,329249,655628,655886,656155,0,0]}}
```

When the user saves the GPIO config, instead of `POST` the form _directly_ to the endpoint, a small bit of JS would `POST` a body to `http://1.2.3.4/api/config/gpio` that looked like ```{"gpio":{"config":[1,1,1,1,1,1,0,0]}}```

-----

As of _right now_ I have found that there is a `http://1.2.3.4/config.json` endpoint which is _very_ helpful but it appears that there are a few issues with using that:

- could go away/move
- The only way to _set_ a password for MQTT/HTTP/WiFi is via `POST` or telnet (or, once MQTT password is set, i guess it and the telnet interface behave the same so you could then set a HTTP password once MQTT connection works...)
 - There is no programmatic way to restart the device w/o first overriding the content of `config.json` with whatever values exist in memory. This means that I can't adjust the `gpio:[]` array in `config.json` and then upload a new `config.json` to the device and expect it to persist after calling `/reboot` I need to emulate the browser and send a `POST` to the `config/gpio` endpoint  to get the new values in memory and _then_ call `/reboot` to get the changes saved
 - Files (`online.cmd`, for example) still need to be uploaded via `/edit` endpoint... But this is not so critical as the file editor and `/edit` endpoint already appear to be largely independent/distinct from the rest of the web UI as it is.

in short: (almost) all device state can be fetched by fetching the `http://1.2.3.4/config.json` file but any _change_ to device state must be done by emulating the web browser or telnet/mqtt. For the purposes of my tool, am 'ignoring' telnet and MQTT because:

- `telnetlib` was not playing nice
-  I would expect that 90% of openHasp users will only ever use HTTP and HTTP is 'default enabled' w/o credentials so it's almost a guarantee that my tool can work after the device has connected to WiFI 

Tl;DR: making web/telnet behave identically would be very helpful :).

nagyrobi, 2022-01-27:
> * upload misc `jsonl` and `cmd` files

all files supported should be included in this list, like `png` or other image formats.

kquinsland, 2022-01-27:
> > * upload misc `jsonl` and `cmd` files
> 
> all files supported should be included in this list, like `png` or other image formats and `zip`.

Ahh good point. I am just mimicking the `upload` button on the web editor so these file types will probably "just work"

dgomes, 2022-01-27:
http://www.bx.psu.edu/~nate/pexpect/pexpect.html

nagyrobi, 2022-01-28:
> The ultimate goal is to flash a device with openHasp and configure WiFi via UART before re-assembling / deploying the device in its intended location.

Note that for this kind or deployment you can also rely on the built-in `user_config_override.h` functionality, which also brings the advantage of predefining your own factory settings for the plate. This way even if for any reason at the location a hardware settings reset happens (flash storage data corruption for example), the factory default settings will still be yours, even with `config.json` lost.

I've learnt this the hard way using Tasmota: they have implemented a kind of weird recovery to factory defaults by power-cycling the device a few times within a time window. I've ran into a power outage caused by the energy company which fell exactly into this pattern - and found myself with about 30 IoT devices built in walls all returned to factory defaults.

kquinsland, 2022-01-28:
> Note that for this kind or deployment you can also rely on the built-in `user_config_override.h` functionality, which also brings the advantage of predefining your own factory settings for the plate.

Absolutely! I am trying to avoid that route specifically to make the tool accessible to other people that don't have a build env set up though.

You can disable the brownout -> reset functionality with Tasmota: https://tasmota.github.io/docs/Commands/#setoption65

nagyrobi, 2022-01-28:
> You can disable the brownout -> reset functionality with Tasmota: https://tasmota.github.io/docs/Commands/#setoption65

Yes I know. Already done that...
I migrate everything to ESPHome as their architecture design is way more adequate for operating multiple dozens of devices. It doesn't really need to store any settings/config at runtime...

fvanroie, 2022-01-29:
Endpoints proposed:
| endpoint | `GET` | `PUT` | `POST` | `DELETE` 
|-----|-----|-----|-----|-----
| /api/info `GET` | ✅ | |
| /api/config/wifi | 🔲  | 🔲 
| /api/config/mqtt | 🔲  | 🔲 
| /api/config/http  | 🔲  | 🔲 
| /api/config/gui | 🔲  | 🔲 
| /api/config/debug  | 🔲  | 🔲 
| /api/config/gpio  | 🔲  | 🔲  | 🔲  | 🔲

kquinsland, 2022-01-29:
I just wanted to leave this here: https://stackoverflow.com/questions/31089221/what-is-the-difference-between-put-post-and-patch

I don't think there is a ton of value in doing a fully pedantic implementation and arguing over which verb should be used for what and if you are updating an existing resource or if the server can only have a single resource of a given type so there is no such thing as "update" only "set _the_ resource to this state" semantics.

TL;DR: I could totally understand removing the `POST` and `DELETE` column and just using `GET` for "give me" and `POST` for "use this" operations.  NO need to complicate things beyond that... even if it's not a "technically correct" implementation :/

beatagerger, 2022-06-24:
hi everybody,
I'm not an electrician, I have no idea about Linux or Raspi, and I'm also not a programmer, like many consumers in general. The main problem is that you have no view of such people, and the description, such as installation ("flash") setup in such language, which a normal consumer hardly understands. (Translation is also scarce, Goggle a disaster)
The wall housing from Zihatec (sold on Amazon by AZ-Delivery) is really nice, you can replace the outdated "thermostats", it would probably also be a hit in terms of sales figures if there were a reasonable program for it.... Everyone knows Alexa but at least Google Home (Google Assistant on the phone) and NODEMCU doesn't start with OpenHAB, but with Arduino!!!!!! So Arduino= IDE, but there is hardly any setup for that. Tasmota is really great, especially because, like IDE or EasyESP, it loads the current version in the web browser!!!!!!!! BUT!!!!!!!
If you install Tasmota (no matter what kind, display or other) you have a problem with "Touch-Mod" and Toch-Mod" is NOT! the same as "Arduitouch"!!!! The GPIO can also be changed with difficulty, because too if ESP32 has 2SPI, Tasmota fails to pack properly!!!!
"AZ-Toch-Mod" has additional free pins for relay and sensors, because who cares about the board temperature on the wall instead of room or outside temperature (no way to define temperature sensors, although there is a connection) It might be a It's a matter of taste, but maybe you should consider writing the programs in such a way that it's also useful(?)
What do you screw a case to the wall for? = To read information, but not in mini format (that's why you buy a 2.8" display) such as the time (big!!!!, analog or digital) + to read temperatures, so the case is in a corresponding format, And if you're curious, you can "turn the pages" and look at other things, such as the outside temperature or the weather forecast, or watering the flowers on the windowsill... There are a few connections for that!!!!!!!!!! !
BUT IT ALL DOESN'T WORK!!!!!!!!!!!! SO HARDLY ANYBODY BUYS IT!!!!!!!!! As long as it stays that way, no one will buy it, stay with the Nextion, and also HomeAssistant (Nouve Casa) which ties everything together. Installation is easy, Alexa can take over the automation through the routines (timer/event etc) I've been trying to get this case to work for a few months, but it's NOT working, at least not to my liking!!!!!!!! ! (NOT the "pages.jasonl" either!) either you have the inputs through Tasmota (Board-ESP32 from AZ-Delivery gets too hot, so something doesn't fit) or you have an empty display (without connections) and the board doesn't work hot.
"override" in gitpod doesn't work either, shows error for "toch-mod" even though it is NOT! ArduiTouch is (as required) The piezze (GPIO 21) is also missing!!!!!!!!
Hompoint, on the other hand, works together with HomeAssistant.

beatagerger, 2022-06-24:
You can delete the kommantera, still it's true!!!!!!!!!
Are you building for yourself or is the release for everyone??????? If it is supposed to be for everyone, then it must also be described accordingly... That is understandable for everyone!!!!!

beatagerger, 2022-06-24:
> Ich wollte das nur hier belassen: https://stackoverflow.com/questions/31089221/what-is-the-difference-between-put-post-and-patch
> 
> Ich glaube nicht, dass es eine Menge Wert hat, eine vollständig pedantische Implementierung durchzuführen und darüber zu streiten, welches Verb für was verwendet werden sollte und ob Sie eine vorhandene Ressource aktualisieren oder ob der Server nur eine einzige Ressource eines bestimmten Typs haben kann, so dass es keine Semantik wie "Aktualisieren" gibt, nur "Setzen Sie _die_ Ressource auf diesen Zustand".
> 
> TL;DR: Ich könnte es total verstehen, die und-Spalte zu entfernen und nur für "gib mir" und für "benutze diese" Operationen zu verwenden. KEINE Notwendigkeit, die Dinge darüber hinaus zu komplizieren ... auch wenn es sich nicht um eine "technisch korrekte" Implementierung handelt :/`POST``DELETE``GET``POST`

The problem I see is that things are not used according to the original intentions... Who cares about the temperature of the board, why can't I install and display a "proper temperature sensor" if there is appropriate space on the board and disk are intended and free? Sensor for "movement" mounted on the wall???? That's ridiculous, instead of DHT, DS18b etc... Tasmota works with GPIOs, but not with the display... So Nextion + IDE remains, or you solder a corresponding board yourself, however, only for one can is the Zichetec ( AZ delivery) too expensive....

nagyrobi, 2022-06-25:
@beatagerger 
You'll have to accomodate to the way it is, as long as it's still _free_ for you.
Don't use this if you don't like it.

fvanroie, 2022-06-25:
@beatagerger Please use the [discussions board](https://github.com/HASwitchPlate/openHASP/discussions) to vent your frustrations or ask questions.

The issue tracker is used to solve verified problems and new features **only**.
The reason your post was deleted is because it has **nothing** to do with [Updates for Responsive web UI](https://github.com/HASwitchPlate/openHASP/issues/246#) #246 and hence does NOT belong here.

If you don't want your posts deleted, please use the appropriate tools and appropriate language! Thank you.

beatagerger, 2022-06-25:
Sorry but what do you mean by "appropriate language"??????????

I wrote in English through Google, like now...

"has nothing to do with update"????
 then what please??????

you install Tasmota in the full version, since this is the prerequisite...... Well, all sensors...... Then update to "AZ-Touch" and apart from a black screen nothing works, not even on the web interface can the relays be switched, sensors have disappeared although they are connected!!!!!

Do you think it has nothing to do with update/upgrade???

In addition, the device sends the temperature value from the board to HomeAssistant via MQTT!?!?! For what reason???? I don't want to know

beatagerger, 2022-06-25:
>[Nagyrobi](https://github.com/nagyrobi)

"for free"????

except IFTT all are free, but all others have correct description, just not here!!!!!

HASmota/Tasmota is offered in many languages, but the translation is a disaster, if you can't even speak a little English, can't even configure the GPIO, the other settings certainly can't... I still don't know how eg . the usual settings, such as "Berry language" or "Command" should be carried out under "az-touch-mod-esp32_ili9341_full_4MB_v0.6.3_497f611" ?????
How can I use my connected relays and sensors here?

fvanroie, 2022-06-25:
@beatagerger This is not Facebook or Twitter but an Issue Tracker for code developers!

**HASmota/Tasmota ** is a completely different project with its _own_ repository and maintainers. Tasmota has nothing to do with openHASP firmware. Again, your comments in this thread are misplaced and will be deleted. Use the appropriate repo and discussion threads provided.

Please visit [https://github.com/tasmota/](https://github.com/tasmota/) for help using Tasmota, Berry and HASPmota.

beatagerger, 2022-06-25:
I haven't used facebook for years.

you chat

You have problems (not just me) and if you don't know the answer, people block you.

For you this is a solution, for us users a "laughing stock"...

don't be surprised if

- no donation

- too few users!!!!

Yes, in the HASSIO (or Arduino) community you are in better hands if there are problems...

You don't want to swear, so "I wish you something" (not nice)

fvanroie, 2022-06-25:
@beatagerger For the 3rd time: you are posting in **wrong** repository for the **wrong** project and you won't accept any help when pointing to the correct information. So, the problem is closed here.

beatagerger, 2022-06-25:
look on the internet!!!!!

we, "other users" (Arduino, tasmota, HASSIO) "discuss" solution for AZ_Touch on many pages......

Your "commentator" advertises on countless pages:

"I suggest using openHASP"

I would have never had an idea because I've never used "openHAB" or Linux ("Pinquin")!!!!!!!!!!!

It was probably caused by the translation, my device is German, everything is translated automatically...

"HASmota" is as you wrote, is not the same as "openHASP"...

your version maybe "change name"???

under Tasmota (Github) finds countless posts, as well as on "Gitpod" ...

take a look...

nielsnl68, 2022-06-25:
Hello beatagerger,

Sorry for bumping in, it is not that we dont want to help you there where a couple of issues with it in your first post that did not  part of this discussion. The second thing is that your post pointed a couple off things at once, that while reading where not understandable.  Also using CAPS is in most cases used for shouting and that is not wanted here.

I understand you have some problems with openHASP. But take a moment, read your first message, go to the discussion section.
Create a new topic there and explain what you are using as display and what steps you have tried and where not able to figure out. 
I agree with you that the documentation could be better, but it is all done in spare time. And it helps when you explain where you get stuck.

Let's stop this overheated discussion here, it is already to warm outside.

commit 1a50f055 (2021-11-16): Minor css tweaks and fixes #246
closed 2023-07-23 by fvanroie