HASwitchPlate/openHASP #250: Add support for ESP32-S2

fvanroie, 2021-11-19:
## Description

There are several new devices coming to the market that use the newer ESP32-S2 and eventually ESP32-S3.
These MCUs are now supported in Arduino core since the v2.0 release.

To limit the number of configurations to test, we will also be removing support for the defunct ESP8266 in the process.

## Tasklist
- [ ] Update framework to Arduino 2.0
- [ ] Update libraries to support ESP32-S2:
  - [ ] TFT_eSPI
  - [ ] LovyanGFX
- [ ] Test ESP32-S2 development boards:
  - [ ] d1 mini ESP32-S2
  - [ ] Adafruit Feather ESP32-S2
  - [ ] Adafruit Metro ESP32-S2
  - [ ] Sparkfun Thing Plus
  - [ ] ATMegaZero ESP32-S2
- [ ] Test off-the-shelf devices:
  - [ ] WT-86-32-3ZW1
  - [ ] ESP32-S2-HMI-DevKit-1

codewise-nicolas, 2021-12-10:
I got a WT-86 device, and in looking to try openHasp on it I came across this persons fork and changes
[https://github.com/paultbarrett/openHASP](https://github.com/paultbarrett/openHASP)
[https://github.com/paultbarrett/openHASP/commit/ad46ec9fbf347a55a5af814772dc760ed86d2064](https://github.com/paultbarrett/openHASP/commit/ad46ec9fbf347a55a5af814772dc760ed86d2064)
I haven't tried it yet, but it may help this issues efforts.

Thanks for the great work fvanroie.

radimkarnis, 2022-01-02:
Would be really nice to see support for the [ESP32-S2-HMI-DevKit-1](https://www.espressif.com/en/news/ESP32-S2-HMI-DevKit-1) by Espressif.

fvanroie, 2022-01-02:
> Would be really nice to see support for the [ESP32-S2-HMI-DevKit-1](https://www.espressif.com/en/news/ESP32-S2-HMI-DevKit-1) by Espressif.

Yes indeed, but seem to be out-of-stock everywhere at the moment... Digikey and Mouser list the device as obsolete now!? Weird.

codewise-nicolas, 2022-01-02:
Would it help your work if you had a WT-86-32-3ZW1 in hand?

I can see about arranging to have one sent to you from my supplier.

fvanroie, 2022-01-02:
Sure, nothing better then hand-on experience.

codewise-nicolas, 2022-01-03:
> Sure, nothing better then hand-on experience.

Send me an email and Ill work out the shipping and such. codewise.nicolas@gmail.com

fvanroie, 2022-01-22:
Thanks to @codewise-nicolas I'm receiving a WT-86-32-3ZW1 test device next week.

fvanroie, 2022-01-29:
Some progress has been made:
[image]

codewise-nicolas, 2022-02-22:
Looks like the pins  for the ESP32S2 are different than the ESP32, so some IO cannot be selected in the GPIO settings. Im working on a PR to fix this. (For example, IO10 and IO11 are brought out to the pads where the voice chip would normally go, making it an easy location to connect something)

Also upped the LV_MEM_SIZE to 30* as even with only a few pages it was running upto the 20* limit. The ESP seems to have plenty of space free, could maybe even go to 40* or 50* if needed.

```
Device Memory
--
Free Heap: | 65.55 KiB
Free Block: | 50.98 KiB
Fragmentation: | 22
PSRam Free: | 1.87 MiB
PSRam Size: | 1.99 MiB

LVGL Memory
--
Total: | 30.00 KiB
Free: | 10.02 KiB
Fragmentation: | 5

```

Fischkopppp, 2022-03-03:
Hi fvanroie, I also have the WT-86 device. I used your user settings from: openHASP/blob/master/user_setups/esp32s2/wt-86-32-3zw1.ini for the User_setup.h file in the TFT_espi library. However it is not working for me. Should it work this way? Have you changed the library to get it working?

codewise-nicolas, 2022-03-04:
The code in the repo should be fully compiling and working. I just did a build a few days ago (as I work to fix the IO thing).
The only edit from the repo is to uncomment the user_setups/esp32s2/*.ini line 11 in the platformio_override.ini and select the proper wt86 environment.
Do you get a compile error?

Fischkopppp, 2022-03-04:
I have not tried openHASP yet. To be honest I am still trying to figure out how to use this repo in platform.io / VS code. I was following this manual: https://openhasp.haswitchplate.com/0.6.1/compiling/local/
However I was not successful yet, but will keep trying to figure out how to use it. As I need the WT86 for my own little project (build on the TFT_espi library) I tried to simply modify the User_Setup.h file according to the wt-86-32-3zw1.ini file. However the screens keeps dark.

```
#define ILI9488_DRIVER     // WARNING: Do not connect ILI9488 display SDO to MISO if other devices share the SPI bus (TFT SDO does NOT tristate when CS is high)

#define TFT_WIDTH  320
#define TFT_HEIGHT 320

#define TFT_BL   0// LED back-light control pin
#define TFT_BACKLIGHT_ON HIGH  // Level to turn ON back-light (HIGH or LOW)

#define TFT_PARALLEL_8_BIT

#define TFT_CS   3  // Chip select control pin (library pulls permanently low
#define TFT_DC   38  // Data Command control pin - must use a pin in the range 0-31
#define TFT_RST  37  // Reset pin, toggles on startup

#define TFT_WR    39  // Write strobe control pin - must use a pin in the range 0-31
#define TFT_RD    2  // Read strobe control pin

#define TFT_D0   36  // Must use pins in the range 0-31 for the data bus
#define TFT_D1   35  // so a single register write sets/clears all bits.
#define TFT_D2   34  // Pins can be randomly assigned, this does not affect
#define TFT_D3   33  // TFT screen update performance.
#define TFT_D4   21
#define TFT_D5   18
#define TFT_D6   17
#define TFT_D7   16

#define SPI_FREQUENCY  40000000
#define SPI_READ_FREQUENCY  20000000
```

fvanroie, 2022-03-04:
> I have not tried openHASP yet.

I'm sorry but this issue is meant for openHASP development purposes only. Please jump onto our [Discord server ](openhasp.haswitchplate.com/discord)or [Github discussions](https://github.com/HASwitchPlate/openHASP/discussions) for general support questions...

> As I need the WT86 for my own little project (build on the TFT_espi library)

We do not use the TFT_eSPI for the WT-86-32-3ZW1 device, so I cannot offer any guidance on using that library in your own project. Please open an issue in the TFT_espi repo instead.

Last time I checked TFT_eSPI [parallel displays were untested with the S2 processor](https://github.com/Bodmer/TFT_eSPI/issues/1348) and our comments got deleted.

fvanroie, 2022-03-22:
@codewise-nicolas I've added a [discussion topic for the WT-86-32-3ZW1](https://github.com/HASwitchPlate/openHASP/discussions/311) hardware. Can you add your experience there too please?

closed 2022-11-07 by fvanroie