HASwitchPlate/openHASP #261: Allow asni codes to be disabled for syslog.

geiseri, 2021-12-23:
If you can make: 
[hasp_debug.cpp:29](https://github.com/HASwitchPlate/openHASP/blob/cacdb34df8e8a039d9cc3de1eb2744b786a6d1f9/src/hasp_debug.cpp#L29) publicly mutable then rsyslog output can disable the ansi codes.  These codes seem to confuse rsyslog.

fvanroie, 2021-12-31:
Thanks for reporting! I've added a checkbox for the ANSI codes to the Debug Settings.

fvanroie, 2022-01-02:
I've added the option to *v0.7.0-dev* test build which is available from the [Actions tab](https://github.com/HASwitchPlate/openHASP/actions). Could you test it please and report if this fixes your issue?

geiseri, 2022-01-04:
Better, the ansi is now fixed.  It seems the syslog format is malformed.  let me investigate, i am using rsyslog on debian here.

fvanroie, 2022-01-04:
Okay thank you. That would not surprise me... I've minimally tested this feature using Visual Syslog Server, so any feedback to improve compatibility/compliance will be greatly appreciated.

geiseri, 2022-01-10:
I was looking at the logging code.  Would you have an issue if I removed the internal version of Arduino-Log (thijse/ArduinoLog) and used the version from platformio?  I was looking at nomis/uuid-syslog but it uses its own logging system that looks pretty new and not used outside of his projects.  Syslog is a handy format for network devices, but the protocol is twitchy as hell once structured text is in play.   How big of a PR would you be willing to entertain if it ends up touching a few files?

fvanroie, 2022-01-11:
The Arduino-Log (thijse/ArduinoLog) used in openHASP is a modified version of the library to be able to log the _same_ message to _different_ streams (Telnet, Console & syslog) at once. So, a remove/replace with the official code will certainly break things. It's not a very big modification, so I could probably backport if the new version has other advantages.

The syslog code I use is based on the ArduinoLog flow, so that it can act as a sort of serial port. If I remember correctly, it was based on [this video](https://www.youtube.com/watch?v=GLoXRyAtytY).

fvanroie, 2022-01-20:
> Better, the ansi is now fixed. It seems the syslog format is malformed. let me investigate, i am using rsyslog on debian here.

If the ANSI is OK, I think we can close this issue. For malformed syslog messages, we can open a new one if needed.

closed 2022-01-22 by fvanroie