HASwitchPlate/openHASP #264: ft2build.h: No such file or directory

DSRocker, 2022-01-01:
In file included from src/hasp/hasp.cpp:34:
lib/lv_lib_freetype/lv_freetype.h:17:10: fatal error: ft2build.h: No such file or directory

******************************************************************
* Looking for ft2build.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:ft2build.h"
* Web  > https://platformio.org/lib/search?query=header:ft2build.h
*
******************************************************************

 #include "ft2build.h"
          ^~~~~~~~~~~~
compilation terminated.
*** [.pio\build\d1-mini-esp32_ili9341\src\hasp\hasp.cpp.o] Error 1
In file included from src/hasp/hasp_font.cpp:4:
lib/lv_lib_freetype/lv_freetype.h:17:10: fatal error: ft2build.h: No such file or directory

******************************************************************
* Looking for ft2build.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:ft2build.h"
* Web  > https://platformio.org/lib/search?query=header:ft2build.h
*
******************************************************************

 #include "ft2build.h"
          ^~~~~~~~~~~~
compilation terminated.
*** [.pio\build\d1-mini-esp32_ili9341\src\hasp\hasp_font.cpp.o] Error 1
=============================================================================== [FAILED] Took 50.96 seconds ===

fvanroie, 2022-01-02:
Please make sure you include `--recursive` when [cloning the repository](https://haswitchplate.github.io/openHASP-docs/0.6.2/compiling/local/#clone-openhasp). lv_lib_freetype is a sub-module and it seems it didn't got initialized properly.

DSRocker, 2022-01-03:
Sorry, in which file?
I tried again and cloned 0.6.2. (not the master) and the compiling worked fine; but after flahing to the ESP32 it blinking an nothing happens...

nagyrobi, 2022-01-03:
What command did you use to make the clone?

DSRocker, 2022-01-03:
I just downloaded the zip-file

nagyrobi, 2022-01-03:
That's not cloning. You're not following the correct steps. Please read carefully:
- https://haswitchplate.github.io/openHASP-docs/0.6.2/compiling/local/
- https://haswitchplate.github.io/openHASP-docs/0.6.2/compiling/gitpod/

DSRocker, 2022-01-03:
thanks! I'll do it correct :-)
another question: by compiling with gitpod I get the bin-file as version v0.7.0.bin

nagyrobi, 2022-01-03:
yes

DSRocker, 2022-01-03:
is the 0.7.0 better than the 0.6.2?! 
I would like to use my display horizontal (not vertical); should I change the values:
tft_rotation = 0      ; default rotation
tft_width = 240       ; default width / to 320
tft_height = 320      ; default height / and here to 240
in the lcd_config.ini

or is it enough to change this in the gui?

nagyrobi, 2022-01-03:
try GUI first!

DSRocker, 2022-01-03:
I just tried to compile again (git submodule update --init --recursive , ...) it's success, but the esp32 LED blinking and nothing happenning...

nagyrobi, 2022-01-03:
OK that's another issue, this can be closed.
I suggest to go to Discorpd support forum https://discord.gg/S8Yrzey2 for faster replies.

closed 2022-01-05 by DSRocker