HASwitchPlate/openHASP #265: WebUI supports mqtt hostnames up to 63 chars in length but only stores the first 32 chars

kquinsland, 2022-01-02:
Hey! Thanks for creating this updated / maintained version of the original HASP project. I had a lot of fun modifying that to add some RGB leds and light sensor and more.  It's a wonderful idea and I am so happy to see that there's now a way to generate graphics on the fly!  This makes things a lot easier to work with.

Please keep up the __amazing__ work!.

-----

### Perform all steps below and tick them with [x]
- [X] Check the related part of the [Documentation](https://haswitchplate.github.io/openHASP-docs/)
- [X] Update openHASP to the latest version
- [X] Reproduce the issue and describe all steps

### Describe the bug

I just flashed the latest release to my WT-32 module and got it connected to WiFi.
I then went to set up MQTT and rebooted the device and was greeted with a `Connecting...` progress bar.

I used the exact same configuration that I know works with Tasmota and am 100% confident that the credentials, host, DNS, MQTT broker ... etc are all OK.

I dig in a bit deeper and notice this on the `config/mqtt?` web endpoint:

```html
<input id="host" required="" name="host" maxlength="63" placeholder="mqttServer" value="mqtt.my-super-lo">
```

The _full_ value of the server is `mqtt.my-super-long-host-name-here.com`.

Looking here:

https://github.com/HASwitchPlate/openHASP/blob/c5d405cafa5e54a1b0328da7ac4c74a6812ac289/src/mqtt/hasp_mqtt_pubsubclient.cpp#L57

It looks like the host buffer is `MAX_USERNAME_LENGTH` long.

And `MAX_USERNAME_LENGTH` appears to be 32 chars long:

https://github.com/HASwitchPlate/openHASP/blob/c5d405cafa5e54a1b0328da7ac4c74a6812ac289/include/hasp_conf.h#L19

I am reasonably confident that the value _should_ be [`MAX_HOSTNAME_LENGTH`](https://github.com/HASwitchPlate/openHASP/blob/c5d405cafa5e54a1b0328da7ac4c74a6812ac289/include/hasp_conf.h#L22).

**Note:** The longest possible value for a hostname is 253 characters, not 128. If there's enough room in flash, I'd suggest bumping max host length up to the full 253 chars or at least putting a `NOTE: ...` somewhere in the `hasp_conf.h` pointing out the potential issue.

### To Reproduce

To reproduce, do the folowing: 

- Set up a MQTT broker that can be accessed via a FQDN of at least 32 characters long.
- Configure openHASP to use this MQTT broker
- Observe the `info?` endpoint and see that the server has only the first 16 chars of the FQDN listed: `mqtt.my-super-lo` and that the status says `Connection failed`

```MQTT
--
Server: | mqtt.home.karlqu
Username: | tasmota
Client ID: | plate01-7b4d10
Status: | Connection failed
Received: | 0
Published: | 0
Failed: | 0
```
### Expected behavior

I would expect that the mechanism to set/store/retrieve the FQDN for a host uses the correct size at every step. 

### Screenshots or video

Apologies if the links provided above are not quite the right place to fix the error / bump up the length. I skimmed the source quickly using GH search.

fvanroie, 2022-01-02:
Thank you for your detailed report!

Yes, there is a mistake in the declaration of the mqttServer array length. It indeed should be:

```cpp
 char mqttServer[MAX_HOSTNAME_LENGTH]   = MQTT_HOST; 
```

A maximum hostname length of 128 was chosen as a trade-off between memory size and usability. Does your superlong hostname fit within 127 characters? If you _really_ need a hostname of up to 253 characters, it can also be overridden in `user_config_override.h` by setting:

```cpp
#define MAX_HOSTNAME_LENGTH 253
```

So far I have yet to encounter these very long DNS hostnames, so I'm curious as to what services uses those long hostnames... In IOT devices, memory is a vary scarce resource and reserving 253 bytes wile not using them would be wasteful.

The current [defaults ](https://github.com/HASwitchPlate/openHASP/blob/09ae100b27f6e3039212f3896ae1db745f7f3f22/include/hasp_conf.h#L15-L23) are:
- Hostnames: 127
- Usernames: 31
- Passwords: 63

fvanroie, 2022-01-02:
I've fixed the issue for *v0.7.0-dev* test build which is available from the [Actions tab](https://github.com/HASwitchPlate/openHASP/actions).

kquinsland, 2022-01-02:
> A maximum hostname length of 128 was chosen as a trade-off between memory size and usability. Does your superlong hostname fit within 127 characters? If you _really_ need a hostname of up to 253 characters, it can also be overridden in `user_config_override.h` by setting:

I figured that was the reasoning. I had completely forgotten that the FQDN will need to _also_ exist in RAM... which is more constrained than flash. My particular hostname does _just_ fit under 64 chars, so the 127 length is _plenty_ for me. I think this will be the case for *most* users but there will always be that one user that relies on the standard a'la https://xkcd.com/1172/.  

> So far I have yet to encounter these very long DNS hostnames, so I'm curious as to what services uses those long hostnames... In IOT devices, memory is a vary scarce resource and reserving 253 bytes wile not using them would be wasteful.

The memory scarcity will (slowly) improve over time... but for _now_ it is a constraint. 
 
I can imagine a hosted MQTT broker with a random/machine-generated FQDN quickly reaching/exceeding the 127 char limit but I don't forsee _most_ openHASP users using a service like that.

E.G:

`some-user-string-here.21c67b0d0735f8a550e6ffb62f87e3f2e27de12d.mqtt.iot.us-west-2-den-1a.prod.someHostedMqttService.tld`

That's 119 chars so still <127 but it's not too difficult to imagine a service that could add some more information to the FQDN which would push it over 127... perhaps if the hash was a sha256 instead of a sha1.

Hopefully a service like that would give the user a CNAME or similar to keep things neater / easier to remember/ type :).

------

> I've fixed the issue for v0.7.0-dev test build which is available from the Actions tab.

I [see that here](
https://github.com/HASwitchPlate/openHASP/actions/runs/1645988950), thanks! I have a _lot_ on my todo list so it's unlikely that i'll be able to set up a build environment this weekend but may be able to do so in the next several days.

closed 2022-01-02 by kquinsland

fvanroie, 2022-01-02:
> It should be noted, however, that the HTML that is sent to the browser _still_ has the hard-coded length of `63` chars. 

The HTML code in my plate is:
```html
<input type="text" id="host" name="host" maxlength="127" placeholder="Server Name" value="">
```

`openHASP/src/sys/svc/hasp_http_async.cpp` is not used, unless you enable the async webserver flag.
By default, we're using the synchronous webserver `openHASP/src/sys/svc/hasp_http.cpp` code.

kquinsland, 2022-01-03:
> Thanks for testing and confirming the issue is resolved.
> 
> > It should be noted, however, that the HTML that is sent to the browser _still_ has the hard-coded length of `63` chars.
> 
> The HTML code in my plate is:
> 
> ```
> <input type="text" id="host" name="host" maxlength="127" placeholder="Server Name" value="">
> ```
> 
> `openHASP/src/sys/svc/hasp_http_async.cpp` is not used, unless you enable the async webserver flag. By default, we're using the synchronous webserver `openHASP/src/sys/svc/hasp_http.cpp` code.

Doh!, you're right. That's what I get for doing a super quick search/skim of the source.