HASwitchPlate/openHASP #273: Long-ish MQTT passwords work but do not survive a reboot

kquinsland, 2022-01-08:
### Perform all steps below and tick them with [x]
- [X] Check the related part of the [Documentation](https://haswitchplate.github.io/openHASP-docs/)
- [X] Update openHASP to the latest version
- [X] Reproduce the issue and describe all steps

### Describe the bug

After a restart, device does not join MQTT, the `/info` endpoint indicates failed auth but does have the correct user name listed.

I can set the password via the `/config/mqtt/` endpoint and the device immediately connects to the MQTT broker.

The device stays connected until reboot.

I tried to check the config file using the file browser and the password has been replaced with `***`... so I don't know what is _really_ being stored on flash.  If there is a way to reveal this, let me know!

### To Reproduce

This is the same wt-32 module from #265. I flashed the build of .7 linked there and went about messing with the module. I left the module and then later had to unplug it.

I plugged the module in, it booted and then jumped to the `Connecting...` screen again. 

I checked the web `/info` endpoint and saw the auth failure message for the MQTT status.

I re-configured the password and the module connected. The module was used for a few moments and I then had to stop working and unplug it.

I came back _again_ after a few hours and plugged it back in, noticed the same issue and was able to solve it by re-supplying the correct password.

Here is a string that is very similar to the password: 

```
deadb3.efdeadbe.efdeadb.e3fde.adbee.fdeadeadb3
```

It is the same length (46 chars) and has a very similar word delimiter pattern.

The password is made up of only alphanumeric characters. I have replaced some of the `e` below with `3` at about the same rate that numbers are used in the real password.

You _should_ be able to reproduce this with the example password above.  If not, let me know.

### Expected behavior

I would expect the correct MQTT password to be used every time the device boots.

### Screenshots or video

N/A

Since the module connects right away once the correct password is supplied, it's not a problem between the browser and web server and MQTT client... so that must mean the flash persistence code... but beyond that, I am not 100% sure how that works.  The JSON libraries for micro controllers are a bit hard to follow sometimes :/.

fvanroie, 2022-01-08:
After changing the password, you need to do a soft-Reboot by using the web UI button or issue the `reboot` command to save the changes to flash.
If you just unplug/power down the device without doing a reboot first, changes are not written to permanent storage.

The `MAX_PASSWORD_LENGTH=64` so that shouldn't be an issue. It looks to me that the changes are not being written due to a hard poweroff.

nagyrobi, 2022-01-08:
@fvanroie maybe we should rename the button to `Save settings / Reboot` so it's straightforward for everyone.

kquinsland, 2022-01-08:
> The `MAX_PASSWORD_LENGTH=64` so that shouldn't be an issue. It looks to me that the changes are not being written due to a hard poweroff.

That would make sense if the values are not persisted as soon as they're used to configure the MQTT client...

> After changing the password, you need to do a soft-Reboot by using the web UI button or issue the `reboot` command to save the changes to flash. If you just unplug/power down the device without doing a reboot first changes are not written to permanent storage.

... Which is sounds like is the case.

I was able to confirm this by using the soft reboot button.  The device does properly connect from a cold power up, now.  I will close this issue out!

> @fvanroie maybe we should rename the button to Save settings / Reboot so it's straightforward for everyone.

This is similar to how Tasmota does it... except that they don't bother changing the text on the button.

This would be a simple fix... but there might even be a simpler fix: A simple `Note: ....` in the footer of every config page so people like me are not confused but could also choose _when_ to reboot. This way, i can use the web to make other changes and queue them up for application/persistence all at once with a single reboot rather than a reboot after each change.  This is _roughly_ equivalent to how the `backlog` command works with Tasmota devices for chaining multiple config changes together before finalizing with a single reboot.

closed 2022-01-08 by kquinsland

fvanroie, 2022-01-08:
> A simple `Note: ....` in the footer of every config page so people like me are not confused but could also choose _when_ to reboot. 

It is in the [documentation](https://haswitchplate.github.io/openHASP-docs/0.6.2/configuration/mqtt/#commandline), but still a fair point... something to be improved upon.

> This way, i can use the web to make other changes and queue them up for application/persistence all at once with a single reboot rather than a reboot after each change.

That's indeed the reasoning behind *not* rebooting automatically. This saves some write cycles to flash by only writing changes on reboot.

> This is _roughly_ equivalent to how the `backlog` command works with Tasmota devices for chaining multiple config changes together before finalizing with a single reboot.

We've opted for the explicit reboot, instead of an implicit one.

> I tried to check the config file using the file browser and the password has been replaced with ***... so I don't know what is really being stored on flash. If there is a way to reveal this, let me know!

All passwords are masked intentionally. Revealing those would be considered a security issue.

fvanroie, 2022-01-10:
[image]