HASwitchPlate/openHASP #278: Expose the display resolution / orientation values easily via device attributes

kquinsland, 2022-01-10:
**Is your feature request related to a problem? Please describe.**

I want to use a simple layout for multiple classes of device. Each device type has a different screen resolution.
I want to define the UI elements via a normal `jsonl` file and then I want to set the X/Y cords of each widget at runtime via MQTT.

E.G.: I want three text labels to float equidistant between each other on the same line. Assuming I know the length of each label, all I need to know is what the width of the screen is to calculate where each one should go.

**Describe the solution you'd like**

If the device can send the display resolution and orientation back to the HA integration then it should be possible to do something like this:

```yaml
plate01:
  objects:
    - obj: "p0b1"
      properties:
        "text": "{{ states('sensor.foo_bar') }}"
        # Allocate 85 pixels for the text
        # We want to be centered in the left third column, check orientation then split in 3, sub 1/2 of length
        "x": >-
            {% if states.openhasp.plate01.orientation in [0,1] %}
              {{states.openhasp.plate01.xres}}3/-(85/2)
            {% else %}
              {{states.openhasp.plate01.yres}}3/-(85/2)
```

**Describe alternatives you've considered**

I considered writing my own script around jinja2 or similar to generate the yaml and or `jsonl` documents that I would use 

**Additional context**

I am trying to get a few "universal" widgets on all screens for a few different openHASP devices.
heading along the top of the screen for time and other short things
tab navigation along the bottom of each screen

The whole idea is that I can compose a few 'generic' `jsonl`file that sets the 'universal' elements to page0 and then i'll add the application specific widgets to  page 1..11.

I don't want to maintain a library of which screens are deployed where and what the resolution is. I want the computer to figure out what the resolution is and then place the widgets at the appropriate location.

fvanroie, 2022-01-10:
This is my proposal for the `disp` discovery key:

```json
{
  "node":"plate01",
  "mdl":"FreeTouchDeck",
  "mf":"openHASP",
  "hwid":"7c87cee34b7c",
  "pages":12,"sw":"0.7.0",
  "uri":"http://10.1.0.133",
  "input":{},
  "power":[],
  "light":[],
  "dim":[],
  "disp":[480,320]
}
```

The orientation should be inferred from the height/width. The display driver specifies a native resolution, but `rotation` doesn't mean portrait or landscape, it depends on the native resolution of the physical panel.

@dgomes Is this format OK for the CC or would it be better to have separate `hres`/`vres` keys?

kquinsland, 2022-01-12:
I can now reliably change the display orientation for a test device and a few seconds after restarting to apply the changes... HA has the correct attributes. This should be enough for my needs!

closed 2022-01-12 by kquinsland

fvanroie, 2022-01-12:
Thanks! I will drop the disp discovery key.