HASwitchPlate/openHASP #282: Add support for buzzer

xtimmy86x, 2022-01-14:
Hello, i'm using this project with a arduitouch that have a integrated buzzer connected to a GPIO of the esp32 board 

It's very interesting if it will be integrated and controllable with the home assistant integration

nagyrobi, 2022-01-14:
Add it as a switch.

xtimmy86x, 2022-01-14:
i did it but no sound, only a soft crackling sound when i put on and off the switch

fvanroie, 2022-01-14:
You need to drive the pitch/length via a library. Take a look at the documentation on how to add your [custom code](https://haswitchplate.github.io/openHASP-docs/0.6.2/compiling/custom/).
Make sure that the code is non-blocking or it will interfere with the performance of the UI and other compoments.

nagyrobi, 2022-01-22:
You're talking about a piezo speaker, not a buzzer. Likely you picked the wrong part.

A buzzer is able to output a sound (predefined from factory) when some DC is applied to its terminals.

A piezo speaker is just a part of the buzzer. It needs some audio signal to be generated and applied to it to act like a buzzer.